FreeCAD/FreeCAD #13268: Sketcher: no perpendicular snap

FEA-eng, 2024-04-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Perpendicular snap (with auto constraint of course) would be really nice. Example from SolidWorks:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-04-02:
This issue is more generic: #11414.

FEA-eng, 2024-04-02:
> This issue is more generic: #11414.

We decided to split it into multiple issues because snaps are implemented independently: https://github.com/FreeCAD/FreeCAD/issues/6240#issuecomment-1987923650

PaddleStroke, 2025-03-26:
I am currently looking into this.
@AjinkyaDahale FYI while doing that I am looking at `DrawSketchHandler::seekAutoConstraint` and will make significant changes to this big function. Just to let you know so that it does not conflict with your current work.

AjinkyaDahale, 2025-03-26:
@PaddleStroke thanks for the info. Please make tests as you refactor, or at least specify the intended behaviour in the tests file so I can.