FreeCAD/FreeCAD #13272: Assembly: fixing individual degrees of freedom

FEA-eng, 2024-04-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I don’t know if the Assembly solver itself already allows this or not (likely not) but, in addition to joints, it would be great to have an option to fix individual degrees of freedom (DOFs) of the selected geometrical features. How would it work? For example, fixing all 3 translational DOFs of a vertex would only allow the rotation of the part around that vertex. Fixing two vertices would allow the rotation around the line connecting those vertices. Properly fixing three vertices would constrain the part completely. The same would be possible with edges and faces and 3 translational directions (global or local X, Y and Z) would be available. Then maybe even rotational DOFs could be added. With proper mouse dragging support (respecting constraints) it would be possible to achieve this unique functionality: https://forum.freecad.org/viewtopic.php?t=70808

@PaddleStroke FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-06:
Not sure if this is easy to implement and I think won't fix as separate Joint type in the UI.
Maybe the Lock Constraint could get a task dialog with options to completely lock (all 6 DOF) or lock only partial movement.

pierreporte, 2024-08-28:
Would it be equivalent to have a generic assembly joint that let the user manually set the 6 degrees of freedom?

FEA-eng, 2024-08-28:
> Would it be equivalent to have a generic assembly joint that let the user manually set the 6 degrees of freedom?

I think so. Assuming that it could be apply to vertices, edges and faces.

kwahoo2, 2025-04-22:
Extending the Lock Constraint sounds good for me. I currently use LC for dragging assemblies with motion controllers, but this can work only if there are all 6 DOFs available, eg. 6 axis robot: https://www.youtube.com/watch?v=VUGKpRUHtX4
For assemblies with lower number of free DOFs I could fill missing ones with DOFs from a custom Lock Constraint.

pierreporte, 2025-12-12:
More than fixing DOFs, it could ask if they are either free or set to a particular value, zero being the equivalent of fixed for that particular DOF.