FreeCAD/FreeCAD #13276: exe clashed when Compiled x64 version with Debug

zhaosky, 2024-04-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I could't find existing issue,so I proposed a new one.when I use msvc19 compiled x64 version with debug,and opened a demo, it crached.
error msg:
> “FreeCAD_d.exe”(Win32): Loaded“D:\Code\FreeCAD-0.21.2\build\bin\Lib\site-packages\PySide2\QtSvg_d.cp38-win_amd64.pyd”。
“FreeCAD_d.exe”(Win32): Loaded“C:\Windows\System32\ExplorerFrame.dll”。
HEAP[FreeCAD_d.exe]: Invalid address specified to RtlValidateHeap( 000001F0865D0000, 000001F0A9165920 )

dump msg:
```
 	ntdll.dll!00007ff86e33690e()	未知
 	ntdll.dll!00007ff86e301805()	未知
 	ntdll.dll!00007ff86e2b6da1()	未知
 	KernelBase.dll!00007ff86b75314b()	未知
 	ucrtbased.dll!00007fffd353e912()	未知
 	ucrtbased.dll!00007fffd353cb51()	未知
 	ucrtbased.dll!00007fffd3540335()	未知
>	Part_d.pyd!operator delete(void * block) 行 38	C++
 	Part_d.pyd!operator delete[](void * block) 行 32	C++
 	Part_d.pyd!NCollection_Array1<int>::~NCollection_Array1<int>() 行 435	C++
 	[外部代码]	
 	Part_d.pyd!Part::Tools::getPointNormals(const TopoDS_Face & theFace, opencascade::handle<Poly_Triangulation> aPolyTri, NCollection_Array1<gp_Dir> & theNormals) 行 579	C++
 	PartGui_d.pyd!PartGui::ViewProviderPartExt::updateVisual() 行 1089	C++
 	PartGui_d.pyd!PartGui::ViewProviderPartExt::onChanged(const App::Property * prop) 行 368	C++
 	PartGui_d.pyd!Gui::ViewProviderPythonFeatureT<PartGui::ViewProviderPart>::onChanged(const App::Property * prop) 行 520	C++
 	FreeCADApp_d.dll!App::Property::hasSetValue() 行 229	C++
 	FreeCADApp_d.dll!App::PropertyBool::setValue(bool lValue) 行 1964	C++
 	FreeCADApp_d.dll!App::PropertyBool::Restore(Base::XMLReader & reader) 行 2004	C++
 	FreeCADApp_d.dll!App::PropertyContainer::Restore(Base::XMLReader & reader) 行 360	C++
 	FreeCADApp_d.dll!App::ExtensionContainer::Restore(Base::XMLReader & reader) 行 308	C++
 	FreeCADGui_d.dll!Gui::ViewProvider::Restore(Base::XMLReader & reader) 行 840	C++
 	FreeCADGui_d.dll!Gui::Document::RestoreDocFile(Base::Reader & reader) 行 1431	C++
 	FreeCADBase_d.dll!Base::XMLReader::readFiles(zipios::ZipInputStream & zipstream) 行 416	C++
 	FreeCADApp_d.dll!App::Document::restore(const char * filename, bool delaySignal, const std::vector<std::string,std::allocator<std::string>> & objNames) 行 2031	C++
 	FreeCADApp_d.dll!App::Application::openDocumentPrivate(const char * FileName, const char * propFileName, const char * label, bool isMainDoc, bool createView, std::vector<std::string,std::allocator<std::string>> && objNames) 行 995	C++
 	FreeCADApp_d.dll!App::Application::openDocuments(const std::vector<std::string,std::allocator<std::string>> & filenames, const std::vector<std::string,std::allocator<std::string>> * paths, const std::vector<std::string,std::allocator<std::string>> * labels, std::vector<std::string,std::allocator<std::string>> * errs, bool createView) 行 779	C++
 	FreeCADApp_d.dll!App::Application::openDocument(const char * FileName, bool createView) 行 671	C++
 	FreeCADApp_d.dll!App::Application::sOpenDocument(_object * __formal, _object * args, _object * kwd) 行 261	C++
 	[外部代码]	
 	FreeCADBase_d.dll!Base::InterpreterSingleton::runString(const char * sCmd) 行 240	C++
 	FreeCADApp_d.dll!App::Application::sLoadFile(_object * __formal, _object * args) 行 229	C++
 	[外部代码]	
 	FreeCADBase_d.dll!Base::InterpreterSingleton::runString(const char * sCmd) 行 240	C++
 	FreeCADGui_d.dll!Gui::Command::_runCommand(const char * file, int line, Gui::Command::DoCmd_Type eType, const char * sCmd) 行 726	C++
 	FreeCADGui_d.dll!Gui::Command::_doCommand(const char * file, int line, Gui::Command::DoCmd_Type eType, const char * sCmd, ...) 行 678	C++
 	WebGui_d.pyd!WebGui::BrowserView::urlFilter(const QUrl & url) 行 607	C++
 	WebGui_d.pyd!WebGui::BrowserView::qt_static_metacall(QObject * _o, QMetaObject::Call _c, int _id, void * * _a) 行 278	C++
 	[外部代码]	
 	WebGui_d.pyd!WebGui::WebEngineUrlRequestInterceptor::interceptRequest(QWebEngineUrlRequestInfo & info) 行 113	C++
 	[外部代码]	
 	FreeCADGui_d.dll!Gui::GUIApplication::notify(QObject * receiver, QEvent * event) 行 85	C++
 	[外部代码]	
 	FreeCADGui_d.dll!Gui::Application::runApplication() 行 2383	C++
 	FreeCAD_d.exe!main(int argc, char * * argv) 行 271	C++
 	FreeCAD_d.exe!WinMain(HINSTANCE__ * __formal, HINSTANCE__ * __formal, char * __formal, int __formal) 行 97	C++
 	[外部代码]	

```

### Full version info

```shell
FreeCAD-0.21.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-18:
@zhaosky can you reproduce with latest dev build ?

wwmmzz, 2024-07-28:
still crash with latest dev 

steps: new a empty project->draft->rectangle
when finish draw, it get crash

dump msg:
```
 	ntdll.dll!00007fff7b8ea592()	未知
 	ntdll.dll!00007fff7b8acffc()	未知
 	ntdll.dll!00007fff7b84e275()	未知
 	KernelBase.dll!00007fff79594a2b()	未知
 	ucrtbased.dll!00007fff1264e852()	未知
 	ucrtbased.dll!00007fff1264ca91()	未知
 	ucrtbased.dll!00007fff12650275()	未知
 	Part_d.pyd!operator delete(void * block) 行 38	C++
 	Part_d.pyd!operator delete[](void * block) 行 32	C++
>	Part_d.pyd!NCollection_Array1<int>::~NCollection_Array1<int>() 行 435	C++
 	[外部代码]	
 	Part_d.pyd!Part::Tools::getPointNormals(const TopoDS_Face & theFace, opencascade::handle<Poly_Triangulation> aPolyTri, NCollection_Array1<gp_Dir> & theNormals) 行 580	C++
 	PartGui_d.pyd!PartGui::ViewProviderPartExt::updateVisual() 行 1110	C++
 	PartGui_d.pyd!PartGui::ViewProviderPartExt::updateData(const App::Property * prop) 行 843	C++
 	PartGui_d.pyd!Gui::ViewProviderPythonFeatureT<PartGui::ViewProvider2DObject>::updateData(const App::Property * prop) 行 295	C++
 	FreeCADGui_d.dll!Gui::ViewProvider::update(const App::Property * prop) 行 309	C++
 	FreeCADGui_d.dll!Gui::ViewProviderDocumentObject::update(const App::Property * prop) 行 383	C++
 	FreeCADGui_d.dll!Gui::Document::slotChangedObject(const App::DocumentObject & Obj, const App::Property & Prop) 行 802	C++
 	[外部代码]	
 	FreeCADGui_d.dll!boost::detail::function::void_function_obj_invoker2<std::_Binder<std::_Unforced,void (__cdecl Gui::Document::*)(App::DocumentObject const &,App::Property const &),Gui::Document *,std::_Ph<1> const &,std::_Ph<2> const &>,void,App::DocumentObject const &,App::Property const &>::invoke(boost::detail::function::function_buffer & function_obj_ptr, const App::DocumentObject & a0, const App::Property & a1) 行 160	C++
 	FreeCADApp_d.dll!boost::function2<void,App::DocumentObject const &,App::Property const &>::operator()(const App::DocumentObject & a0, const App::Property & a1) 行 770	C++
 	FreeCADApp_d.dll!boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::m_invoke<boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)>,0,1,App::DocumentObject const &,App::Property const &>(boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)> & func, boost::signals2::detail::unsigned_meta_array<0,1> __formal, const std::tuple<App::DocumentObject const &,App::Property const &> & args, void * __formal) 行 106	C++
 	FreeCADApp_d.dll!boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::operator()<boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)>,App::DocumentObject const &,App::Property const &,2>(boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)> & func, const std::tuple<App::DocumentObject const &,App::Property const &> & args, boost::mpl::size_t<2> __formal) 行 90	C++
 	FreeCADApp_d.dll!boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type,App::DocumentObject const &,App::Property const &>::operator()<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<enum boost::signals2::detail::slot_meta_group,boost::optional<int>>,boost::signals2::slot<void __cdecl(App::DocumentObject const &,App::Property const &),boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)>>,boost::signals2::mutex>>>(const boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<enum boost::signals2::detail::slot_meta_group,boost::optional<int>>,boost::signals2::slot<void __cdecl(App::DocumentObject const &,App::Property const &),boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)>>,boost::signals2::mutex>> & connectionBody) 行 133	C++
 	FreeCADApp_d.dll!boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type,App::DocumentObject const &,App::Property const &>,std::_List_iterator<std::_List_val<std::_List_simple_types<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<enum boost::signals2::detail::slot_meta_group,boost::optional<int>>,boost::signals2::slot<void __cdecl(App::DocumentObject const &,App::Property const &),boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)>>,boost::signals2::mutex>>>>>,boost::signals2::detail::connection_body<std::pair<enum boost::signals2::detail::slot_meta_group,boost::optional<int>>,boost::signals2::slot<void __cdecl(App::DocumentObject const &,App::Property const &),boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)>>,boost::signals2::mutex>>::dereference() 行 110	C++
 	FreeCADApp_d.dll!boost::iterators::iterator_core_access::dereference<boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type,App::DocumentObject const &,App::Property const &>,std::_List_iterator<std::_List_val<std::_List_simple_types<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<enum boost::signals2::detail::slot_meta_group,boost::optional<int>>,boost::signals2::slot<void __cdecl(App::DocumentObject const &,App::Property const &),boost::function<void __cdecl(App::DocumentObject const &,App::Property const &)>>,boost::signals2::mutex>>>>>,boost::signals2::detail::connection_body<std::pair<enum boost::signals2::detail::slot_meta_group,boost::optional<int>>,boost::signals2::slot<void __cdecl(App::DocumentO [... cut ...]
 	FreeCADGui_d.dll!Gui::GUIApplication::notify(QObject * receiver, QEvent * event) 行 83	C++
 	[外部代码]	
 	FreeCADGui_d.dll!`anonymous namespace'::tryRunEventLoop(Gui::GUISingleApplication & mainApp) 行 2067	C++
 	FreeCADGui_d.dll!`anonymous namespace'::runEventLoop(Gui::GUISingleApplication & mainApp) 行 2083	C++
 	FreeCADGui_d.dll!Gui::Application::runApplication() 行 2162	C++
 	FreeCAD_d.exe!main(int argc, char * * argv) 行 301	C++
 	FreeCAD_d.exe!WinMain(HINSTANCE__ * __formal, HINSTANCE__ * __formal, char * __formal, int __formal) 行 97	C++
 	[外部代码]	
```

closed 2025-07-04 by github-actions[bot]