FreeCAD/FreeCAD #13279: Transformation de Fichier importé WRL

Turban-dominique, 2024-04-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,
on freecad 0.22.0Dev 36731
When I import a *.WRL file, it is correctly placed in the tree, but I can't move it using the transform move tool, whereas all the other imports move fine.
Thank you for your advice.
Dominique 
Bonjour,
sur freecad 0.22.0Dev 36731
Quand j'importe un fichier *.WRL il se place bien dans l'arborescence mais on ne peut pas le déplacer par l'outil transformer déplacer alors que toutes les autres importations se déplace biens.
merci de vos avis.
Dominique 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * A2plus 0.4.62
  * fasteners 0.5.2
  * frame 0.1.1
  * sheetmetal 0.2.63
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-06-22:
Still relevant

FEA-eng, 2026-09-10:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```