FreeCAD/FreeCAD #13298: [Problem] PartDesign_Groove and PartDesign_Revolution: issues with task panel and order of properties.

Roy-043, 2024-04-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. The task panel of PartDesign_Groove has the wrong text and icon: "Revolution parameters".
2. The Face option is always displayed. With 4 out of 5 Type options it is disabled. This is inconsistent when compared to the PartDesign_Pad task panel, but also when you look at the 2nd angle input (which is hidden when it does not apply).
3. I do not understand why the Face button and input box next to it are different compared to PartDesign_Pad. The tools have an obvious relation and should present the same UI here.
4. The 2nd angle input should come after the Angle input.
5. The properties are not ordered A-Z, I don't understand the reason for this. But what is stranger is that the Groove prop group has a different order than the Revolution prop group, even though the properties are the same.
6. The Base property has an awkward name. "Base" in FreeCAD usually refers to an object.
7. The Base and Axis properties are read only. Perhaps they should be hidden?

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-06:
@AjinkyaDahale FYI.

maxwxyz, 2024-07-01:
@FlachyJoe FYI

AjinkyaDahale, 2024-08-21:
So for 1.0 it seems obviously too late to change the UI. If not, please let me know anything that needs to be addressed.

I guess the biggest issue here is the property name "base". What is the term used in the rest of PD? If we need to change it that may be a major issue post release.

The fixes to the other points mentioned should be relatively straightforward, but my concern is that UI issues will only keep coming up more and more so long as everything is written at this low a level. Perhaps a more declarative approach would be useful, like qt-designer telling what's visibility is related to what. If anyone has a suggestion I'd like to hear it.

Roy-043, 2024-09-17:
I see now that the PartDesign helix features use the same property names (Axis and Base). And that "Base" is not used very consistently in FreeCAD. Actually "Base" may refer to a point more often than to an object.

- Part_Extrude uses Base for the sketch.
- Part_Revolve uses Source for the sketch, and Axis and Base as the mentioned PartDesign tools.
- Draft arrays use Base for the object that is arrayed.
- Draft_Shape2DView use Base for the object that is projected.

Let's disregard the property name issue.

FEA-eng, 2025-06-22:
Still relevant

AjinkyaDahale, 2026-09-24:
Anything left in this issue?

Roy-043, 2026-09-25:
The issues has all been addressed AFAICT. Thank you.

closed 2026-09-25 by Roy-043