FreeCAD/FreeCAD #13300: Crash

jeff-at-theranch, 2024-04-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Crash, crash report attached 
[crash.txt](https://github.com/FreeCAD/FreeCAD/files/14874809/crash.txt)

### Full version info

```shell
[code]
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-04-04:
@jeff-at-theranch we need steps to reproduce the crash. Is there a file you can attach that shows the crash occur ?

maxwxyz, 2024-05-25:
Closing for no feedback.

closed 2024-05-25 by maxwxyz