FreeCAD/FreeCAD #13303: Addon Manager refuses to open because of some dependency issue. (module 'PySide.QtWebEngineWidgets' has no attribute 'QWebEngineSettings')

BurntVoxel, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Problem: The addon manager started to refuse to load after the last update. The Report View showed a few errors. 

After freshly reinstalling and making sure dependencies were up to date, this error is consistent: 
```
17:51:53  Traceback (most recent call last):
17:51:53    File "/usr/lib/freecad/Mod/AddonManager/AddonManager.py", line 206, in launch
17:51:53      self.packageDetails = PackageDetails(self.dialog)
17:51:53                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
17:51:53    File "/usr/lib/freecad/Mod/AddonManager/package_details.py", line 83, in __init__
17:51:53      self.ui.setupUi(self)
17:51:53    File "/usr/lib/freecad/Mod/AddonManager/package_details.py", line 863, in setupUi
17:51:53      self.webView.setPage(RestrictedWebPage(PackageDetails))
17:51:53                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
17:51:53    File "/usr/lib/freecad/Mod/AddonManager/package_details.py", line 725, in __init__
17:51:53      QtWebEngineWidgets.QWebEngineSettings.ErrorPageEnabled, False
17:51:53      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
17:51:53  AttributeError: module 'PySide.QtWebEngineWidgets' has no attribute 'QWebEngineSettings'
```
I'm trying to figure out where the exact issue is, it's taking a bit though. Not familiar with PySide.

### Full version info

```shell
[code]
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.8, Qt 6.6.3, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
[/code]
```
// EndeavourOS
```

### Subproject(s) affected?

Addon Manager

### Anything else?

I *think* this part of the QT manual is relevant? https://doc.qt.io/qtforpython-6/PySide6/QtWebEngineCore/QWebEngineSettings.html

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-04-05:
To my knowledge, it's impossible to compile _FreeCAD_ 0.21.2.33771 against Qt 6.6.3 unless you heavily change the source code.

BurntVoxel, 2024-04-05:
> To my knowledge, it's impossible to compile _FreeCAD_ 0.21.2.33771 against Qt 6.6.3 unless you heavily change the source code.

Well... That's how it's published on Arch Linux, requiring Qt 6... So I guess the dependency list needs changed until someone can patch that.

xtemp09, 2024-04-05:
If you look here, [^1] _QWebEngineSettings_ of Qt 5 is a part of _webenginewidgets_, while if you look here,[^2]
_QWebEngineSettings_ of Qt 6 is a part of _webenginecore_.

The error shows that _FreeCAD_ 21.2 tries to set an attribute in the way it supposed to be in Qt 5, while it has been compiled against Qt 6.

The problem is solved. _FreeCAD_ has been compiled against the wrong version of Qt.

[^1]: https://doc.qt.io/qt-5/qwebenginesettings.html
[^2]: https://doc.qt.io/qt-6/qwebenginesettings.html

K1ngjulien, 2024-04-05:
can the latest develop be built against Qt6?

I'm guessing then that anything publishing 21.2 should stick with Qt5 until the next release, right?

K1ngjulien, 2024-04-05:
cool, thanks for the info!

@antonio-rojas seems to be the one who submitted the change to the PKGBUILD.
pls change it back when you see this :pray:

K1ngjulien, 2024-04-05:
@BurntVoxel In the meantime, you can get back the old version by building the package yourself from source:

```bash
git clone https://gitlab.archlinux.org/archlinux/packaging/packages/freecad.git
cd freecad
git checkout 0.21.2-6
makepkg --syncdeps
makepkg --install
```

that should get you back to the old version. might take a while to build but it worked for me :)

antonio-rojas, 2024-04-05:
Should be fixed in `0.21.2-8`

K1ngjulien, 2024-04-05:
@antonio-rojas Addon-Manager works again, but the issues in #13312 still remain... i can't even use the navigation cube in the top left :sweat_smile:

antonio-rojas, 2024-04-05:
> @antonio-rojas Addon-Manager works again, but the issues in #13312 still remain... i can't even use the navigation cube in the top left 😅

Please open a report on the Arch bug tracker (so we stop spamming upstream) and post steps and sample files to reproduce.

BurntVoxel, 2024-04-05:
> > @antonio-rojas Addon-Manager works again, but the issues in #13312 still remain... i can't even use the navigation cube in the top left 😅
> 
> Please open a report on the Arch bug tracker (so we stop spamming upstream) and post steps and sample files to reproduce.

Sorry. Didn't know. I'll look.

K1ngjulien, 2024-04-05:
> Please open a report on the Arch bug tracker (so we stop spamming upstream) and post steps and sample files to reproduce.

Will do, I have requested an account there.

Any particular reason for the arch build moving to Qt6? Will arch drop Qt5?

appreciate the patch attempts, wouldn't it be better to wait until the next release before changing build libs?
There's no way for addons to handle this if the compatibility break happens like this.

> The error shows that FreeCAD 21.2 tries to set an attribute in the way it supposed to be in Qt 5, while it has been compiled against Qt 6.
> The problem is solved. FreeCAD has been compiled against the wrong version of Qt.

antonio-rojas, 2024-04-05:
> Any particular reason for the arch build moving to Qt6? Will arch drop Qt5?

No. It will drop pyside2.

BurntVoxel, 2024-04-05:
> > Any particular reason for the arch build moving to Qt6? Will arch drop Qt5?
> 
> No. It will drop pyside2.

Is this something a volunteer could assist with? I've been meaning to learn more about how this stuff works, and maybe searching for API calls to update would be within my ability.
Catch is, I'm not sure which fork/branch/repo/whatever I'm supposed to be looking at now.

maxwxyz, 2024-04-06:
Not sure if related: https://github.com/FreeCAD/FreeCAD/issues/13118

mrmeowski, 2024-04-10:
Arch Linux bug for navigation cube: https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues/3.

adrianinsaval, 2024-04-22:
The navigation cube thing is a genuine issue in FreeCAD dev, not archlinux packaging related. I have experienced on my own build

adrianinsaval, 2024-04-22:
@maxwxyz if you have time and something like this falls under the scope of your work you think you can make separate issues for each specific problem related to qt6? when they are all lumped together like this and just in comments it's hard to keep track. We could have a separate issue where we have an overview linking all qt6 problems (#6992 was an old attempt) or maybe a project

adrianinsaval, 2024-04-22:
@antonio-rojas you might want to include https://github.com/FreeCAD/FreeCAD/pull/13554/commits/ac9a88c2c59a2d2002a7740a1597a05c819220e0 from #13554 as a patch too, should fix https://github.com/FreeCAD/FreeCAD/issues/13303#issuecomment-2042754575 and https://github.com/FreeCAD/FreeCAD/issues/13533

antonio-rojas, 2024-04-23:
> @antonio-rojas you might want to include [ac9a88c](https://github.com/FreeCAD/FreeCAD/commit/ac9a88c2c59a2d2002a7740a1597a05c819220e0) from #13554 as a patch too, should fix [#13303 (comment)](https://github.com/FreeCAD/FreeCAD/issues/13303#issuecomment-2042754575) and #13533

Done, thanks.

G-Pereira, 2024-04-25:
@antonio-rojas sorry to sapm here but your latest changes are breaking again the addon manager and the opening of complex assemblies (probably some addon it is not up to date)

adrianinsaval, 2024-04-25:
> @antonio-rojas sorry to sapm here but your latest changes are breaking again the addon manager and the opening of complex assemblies (probably some addon it is not up to date)

this is not enough information to do anything. What errors? keep in mind that addons are the responsability of their respective maintainers

G-Pereira, 2024-04-26:
@antonio-rojas I know it isn't of much help but Freecad hangs and give me this error: https://pastebin.com/jFb9a7PQ

G-Pereira, 2024-04-26:
The fact that FreeCAD 0.21.2 has qt6 as a dependency is wrong because freecad is currently having issues with qt6, can you revert it to qt5?

G-Pereira, 2024-04-26:
@antonio-rojas Also, while trying to open Addon manager I get this:
```
18:06:43  During initialization the error "'PySide6.QtWidgets.QStatusBar.removeWidget' called with wrong argument types:
  PySide6.QtWidgets.QStatusBar.removeWidget(QPushButton)
Supported signatures:
  PySide6.QtWidgets.QStatusBar.removeWidget(PySide6.QtWidgets.QWidget)" occurred in /usr/lib/freecad/Mod/Tux/InitGui.py
18:06:43  Please look into the log file for further information
18:06:46  FreeCAD version (currently 0.21.0 (2)) must be at least 0.21.2 (33772) in order to work with Python 3.11 and above
18:06:49  Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/freecad/bin/qtwebengine_dictionaries'
18:06:52  Running the Python command 'Std_AddonMgr' failed:
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/AddonManager/AddonManager.py", line 161, in Activated
    NetworkManager.InitializeNetworkManager()
  File "/usr/lib/freecad/Mod/AddonManager/NetworkManager.py", line 655, in InitializeNetworkManager
    AM_NETWORK_MANAGER = NetworkManager()
                         ^^^^^^^^^^^^^^^^
  File "/usr/lib/freecad/Mod/AddonManager/NetworkManager.py", line 171, in __init__
    self.QNAM.setCache(self.diskCache)

'PySide6.QtNetwork.QNetworkAccessManager.setCache' called with wrong argument types:
  PySide6.QtNetwork.QNetworkAccessManager.setCache(QNetworkDiskCache)
Supported signatures:
  PySide6.QtNetwork.QNetworkAccessManager.setCache(PySide6.QtNetwork.QAbstractNetworkCache)
```

antonio-rojas, 2024-04-26:
> The fact that FreeCAD 0.21.2 has qt6 as a dependency is wrong because freecad is currently having issues with qt6, can you revert it to qt5?

No, as already explained in https://github.com/FreeCAD/FreeCAD/issues/13303#issuecomment-2040642936

You will have to install it some other way if you rely on the broken features.

xyproto, 2024-05-13:
Fedora have removed `pyside2` and are also having issues packaging FreeCAD (until Qt6 and Python 3.12 are supported):

https://github.com/FreeCAD/FreeCAD/issues/13983#issuecomment-2106303047

Jookia, 2024-06-21:
Is there much value in keeping this opened? As far as I can tell everything in this thread has been fixed in the dev build and other Qt6 issues are open in their own issues.

K1ngjulien, 2024-06-26:
yes i think it's worth keeping it open for other users to find.
personally, I'd wait until this is mostly fixed downstream in the repositories, which probably won't happen until 0.22 is released.

maxwxyz, 2024-07-01:
Closed as not actionable. If there still are existing issues, please create individual ones.

luzpaz, 2024-07-31:
Can we create separate tickets for any remaining open issues mentioned in this ticket ? Volunteers ?

luzpaz, 2024-10-20:
What's left to do here ?

K1ngjulien, 2024-10-20:
from trying the 1.0rc2 app image this seems to be fixed. migration away from python2's old libraries is done no?

closed 2024-10-20 by maxwxyz