FreeCAD/FreeCAD #13306: FEM Pipeline 'NoneType' object has no attribute 'Visibility'

FEA-eng, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While testing a model, I've noticed a strange issue with the results pipeline. When I submit an analysis, only the CCX_Results object appears and I get the following errors:

```
11:20:46  Running the Python command 'FEM_SolverRun' failed:
Traceback (most recent call last):
  File "E:\Program Files\FreeCAD_weekly-builds-36731-2024-04-01-conda-Windows-x86_64-py311\Mod\Fem\femcommands\commands.py", line 1174, in Activated
    run_fem_solver(self.selobj)
  File "E:\Program Files\FreeCAD_weekly-builds-36731-2024-04-01-conda-Windows-x86_64-py311\Mod\Fem\femsolver\run.py", line 108, in run_fem_solver
    fea.run()  # standard, no working dir is given in solver
    ^^^^^^^^^
  File "E:\Program Files\FreeCAD_weekly-builds-36731-2024-04-01-conda-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 688, in run
    self.load_results()
  File "E:\Program Files\FreeCAD_weekly-builds-36731-2024-04-01-conda-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 808, in load_results
    self.load_results_ccxfrd()
  File "E:\Program Files\FreeCAD_weekly-builds-36731-2024-04-01-conda-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 818, in load_results_ccxfrd
    importCcxFrdResults.importFrd(
  File "E:\Program Files\FreeCAD_weekly-builds-36731-2024-04-01-conda-Windows-x86_64-py311\Mod\Fem\feminout\importCcxFrdResults.py", line 208, in importFrd
    pipeline_visibility = pipeline_obj.ViewObject.Visibility
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

'NoneType' object has no attribute 'Visibility'
```

I can't add the pipeline manually either.

[Pipeline_issue.zip](https://github.com/FreeCAD/FreeCAD/files/14882841/Pipeline_issue.zip)

@marioalexis84 and @lyphrowny FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-05:
Weird, it seems to be session-specific. It happened after I tried deleting all analysis/results files and got "no memory" error message from the Warp filter.

marioalexis84, 2024-04-09:
@FEA-eng can you provide a step by step to reproduce the project?

marioalexis84, 2024-04-09:
The warp object is ok , but I need to manually add it to the Filter property of the pipeline, and it seems to be created with other constraint parameters since I can't reproduce the same result by running the solver again.
[image]

Also, there is something wrong with visibility as when the file is opened, the Visibility property is False, but in the tree state for the pipeline and warp are marked as visible. I need to toggle the visibility several times to synchronize the property and tree state.

FEA-eng, 2024-04-09:
@marioalexis84 Thanks for checking. Apparently, it's not related to the model setup but rather it's something that happened in my session when I deleted multiple results objects at once. I got the "no memory" error first and then I couldn't obtain the results anymore ever after multiple attempts. Only restarting FreeCAD helped. I was hoping that those error messages could be sufficient to diagnose this issue because it can be hard to recreate it. I think that I had a similar problem (at least with "no memory") when deleting the results objects once with a completely different model. So maybe we shouldn't focus on this specific model but rather try to check the pipeline object deletion in various cases as it may lead to errors. If it's not reproducible at all then we can close the issue but maybe there is a way to figure out what can go wrong wrong.

marioalexis84, 2024-04-09:
Ok, we need to play with the pipeline and try to reproduce the error.

luzpaz, 2024-07-31:
Any progress here ?

maxwxyz, 2024-11-09:
@FEA-eng is this still an issue?

FEA-eng, 2024-11-11:
> @FEA-eng is this still an issue?

Yes, pretty nasty one.

FEA-eng, 2025-10-19:
Still no reliable way to reproduce it