FreeCAD/FreeCAD #13308: [Issue] TopoNaming Phase 3 - import SubShapeBinder/Part Workbench

CalligaroV, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In RT fork some methods in Part and PartDesign Workbenches calls `Part::SubShapeBinder::import()`.
This method, as well as the entire `Part::SubShapeBinder` class aren't present in main.

To make the methods in Part and PartDesign Workbenches work with other code imported for the Toponaming Project it is necessary to import the `Part::SubShapeBinder` class.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36777 (Git)
Build type: Debug
Branch: main
Hash: 7a44179529920e79d40a7f4fe3dd49a48072363f
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * Assembly4 0.50.12
  * OpticsWorkbench 1.0.14
  * ExplodedAssembly
  * Beltrami 1.0.8-alpha
  * ose-piping-workbench
  * FreeCAD_SH2_Panel_Workbench
  * sheetmetal 0.4.10
  * Curves 0.6.29
  * fasteners 0.5.15
  * fcgear 1.0.0
  * kicadStepUpMod 11.1.0
  * LCInterlocking
  * OpenDark 2023.12.17
  * ose-3d-printer-workbench
  * Render 2024.1.23
  * Ondsel-Lens 2024.3.26.01
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-01:
@CalligaroV are you planning to write this PR?

chennes, 2024-08-15:
I'm going to close this as "Not planned" since it doesn't seem like we need/want this chunk of code right now.

closed 2024-08-15 by chennes