FreeCAD/FreeCAD #13310: [Issue] TopoNaming Phase 3 - Update onSelectionChanged()/Sketcher Workbench/DrawSketchHandlerExternal

CalligaroV, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Related to #10402.

In RT fork the Sketcher external geometries are added using `attachExternal()` or `addExternal()` while in main there's only `addExternal()`.

To make the external geometry work with other code imported for the Toponaming Project it is necessary to import also the `attachExternal()` method (in C++ and Python) and update both `addExternal()` and `SketcherGui::DrawSketchHandlerExternal::onSelectionChanged()` methods.

Before this, #13309  must be solved

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36777 (Git)
Build type: Debug
Branch: main
Hash: 7a44179529920e79d40a7f4fe3dd49a48072363f
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * Assembly4 0.50.12
  * OpticsWorkbench 1.0.14
  * ExplodedAssembly
  * Beltrami 1.0.8-alpha
  * ose-piping-workbench
  * FreeCAD_SH2_Panel_Workbench
  * sheetmetal 0.4.10
  * Curves 0.6.29
  * fasteners 0.5.15
  * fcgear 1.0.0
  * kicadStepUpMod 11.1.0
  * LCInterlocking
  * OpenDark 2023.12.17
  * ose-3d-printer-workbench
  * Render 2024.1.23
  * Ondsel-Lens 2024.3.26.01
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-25:
Is this issue still relevant after the TNP fixes?

CalligaroV, 2024-06-01:
Hi @maxwxyz.

Probably at some point we'll need to import the related code.

By now I opened a PR (#14437) that should fix the issue that made me open this issue (#13009) also with TNP code enabled by default.
If that PR goes well we can either close this issue and the related ones or leave it open as a reminder for future work.

maxwxyz, 2024-06-24:
@CalligaroV is this fixed now?

chennes, 2024-08-15:
I'm going to close this as "Not planned" since it doesn't seem like we need/want this chunk of code right now.

closed 2024-08-15 by chennes

CalligaroV, 2024-08-16:
Hi @maxwxyz and @chennes.

the following comment is valid both for this PR, #13309 and #13308:

I agree about closing them. These 3 PRs are related to an analysis I made some moths ago that made me think that the TNP code wouldn't work without them (especially without #13308).

At the current state it looks that my analysis was wrong, therefor I don't see the need for none of those ATM.