FreeCAD/FreeCAD #13312: Broken Arch Build using QT6, pyside6 and/or shiboken6

K1ngjulien, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi!

Yesterday, the archlinux build got updated to 0.21.2-7, which updated to QT6, pyside6 and shiboken6
(see: https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/commit/15263d56dbd4adb0e3e227fd7b534e731c05522b )

I updated, and now FreeCAD is barely useable.
The orientation cube in the corner is not clickable.
If I open anything with more than one part visible its really slow and it freezes for a few seconds after a face has been highlighted with mouse over, with cpu spiking to ~70%

I thought it might be an outdated plugin, but I can't update since the Addon Manager doesn't open anymore...
Assembly 4 might be the cause of the `PySide2/__init__.py: Unable to import shiboken2` error.

Logs are attached below.

I've manually built the old 0.21.2-6 release that uses qt5 and it works normally.

Sorry if this is the wrong place to report this, but from looking at the code it seems those library versions should be supported.

Cheers,
K1ngjulien

(cc: @antonio-rojas )

### Full version info

```shell
[code]
OS: EndeavourOS (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.8, Qt 6.6.3, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * fasteners 0.5.15
  * freecad.gears 1.2.0
  * Assembly4 0.50.12
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/14884171/FreeCAD.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

K1ngjulien, 2024-04-05:
Ok the `PySide2/__init__.py: Unable to import shiboken2` error comes from the `Help` addon. with it manually deleted, the error on startup goes away.

xtemp09, 2024-04-05:
This issue is seriously connected to #13303.

K1ngjulien, 2024-04-05:
> This issue is seriously connected to #13303.

oh i'm very sorry about that, i don't know how i missed that issue :sweat_smile:

maxwxyz, 2024-04-06:
I'll close this then

closed 2024-04-06 by maxwxyz