FreeCAD/FreeCAD #13313: Sketcher: creating external geometry from faces

FEA-eng, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, when clicking on a face when using the External geometry tool, the following error messages appear:

```
13:48:11  Selected external reference plane must be normal to sketch plane
13:48:11  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to add external shape element Face6
13:48:11  Unnamed#Sketch001: App.getDocument('Unnamed').getObject('Sketch001').addExternal("Pad","Face6")
13:48:11  Unnamed#Sketch001: Failed to add external geometry
```

Instead, all the edges of that face should be projected instead of having to click on each one of them.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-05:
Depends on the direction of the face. If the face is perpendicular to the sketch plane it works, but there is this other issue then: https://github.com/FreeCAD/FreeCAD/issues/11824

PaddleStroke, 2024-10-25:
That should be easy to do. I'm on it

FEA-eng, 2024-10-25:
> That should be easy to do. I'm on it

Thanks. IMO it’s really important for usability, especially once we can turn the external geometry into real one.

pierreporte, 2024-11-29:
@PaddleStroke is this solved by #17736?

PaddleStroke, 2024-11-29:
Yes

pierreporte, 2024-11-29:
@maxwxyz can you close it?