FreeCAD/FreeCAD #13314: Sketcher: scale sketch on first dimension

FEA-eng, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It has been requested on the forum but I haven't found any GitHub issue for this. It would be very useful to have an option to scale the whole sketch after entering the first dimension. This way one could draw the initial shape and dimension it without worrying about it braking because the first dimension is much bigger/smaller than the current size of the geometry. Such a feature is available in SolidWorks, for instance.

Forum discussion: https://forum.freecad.org/viewtopic.php?t=58172

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2025-04-08:
I need this badly, as it ruines my normal workflow.
It is so bad that I very often spend more time in fixing my ruined sketch instead of making the actual sketch.

theo-vt, 2025-04-21:
Hi @FEA-eng  and @MisterMakerNL , I made a small prototype, that works like so:

[image]

Is it how you would expect this feature to work?

Also, how would the undo stack work with this feature, should the scaling be within the same command as setting the dimension?

Thanks!

MisterMakerNL, 2025-04-22:
Wow that is awesome!
For the undo, I think it should just go back before you input the size in the dimension so after you place the dimension, if that is possible.

hyarion, 2025-04-22:
> Is it how you would expect this feature to work?

The view "jumps" a bit when you add that first dimension. Would it be possible to adjust the zoom so the geometry stay in the exact same place?

Panda631, 2025-04-22:
There are so many good ideas in the forum, but most of them are forgotten over time.

theo-vt, 2025-04-23:
Hi hello, 

I have changed the way the camera is adjusted and now the geometry is static, however the annotation moves out of the way and I am not sure how to fix it,

https://github.com/user-attachments/assets/7c256b59-be52-4459-a445-0c34c43b38ae

Thanks!

MisterMakerNL, 2025-04-24:
@theo-vt I really don't care to much about the small details/bugs, it works good enough for me. And I love it😍

hyarion, 2025-04-24:
This is a great start!
But without adjusting the positions of the dimension so it follows the geometry, you now need to zoom to position the dimension in the view.

If you push your code somewhere I could take a look and see if I can help you.

theo-vt, 2025-04-24:
@hyarion it's up at: https://github.com/theo-vt/FreeCAD/tree/sketch_autoscale

I'll try to find where dimension positions are set on my side as well

Thanks!

theo-vt, 2025-04-30:
Hi I think I figured how to prevent the annotation from jumping all over the place. 

https://github.com/user-attachments/assets/cdfc1635-e824-46ca-84e1-f6b01329f334

There are still some weirdness with the zoom when the camera is not centered which I need to investigate,

FlachyJoe, 2025-04-30:
> it's up at: https://github.com/theo-vt/FreeCAD/tree/sketch_autoscale

Please open a (draft) PR so we can easily pull and test your branch and comment.

theo-vt, 2025-04-30:
Oh I was not aware that it was how it's done, thanks!

It is now draft PR #21084

PR #21084 (2025-06-09): Sketch autoscale

Triggers a scale action when there is a single dimensional constraint in a sketch and it's datum is edited to avoid breaking the overall shape of the sketch.

## Issues

fixes #13314