FreeCAD/FreeCAD #13316: Sketcher: a button to hide all construction geometries and constraints

FEA-eng, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There's no easy way to hide all construction geometries and constraints when editing a sketch. Such a button could be very useful.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-02:
@FreeCAD/design-working-group should there be a single button to hide all construction and internal geometry plus all constraints or just the constraints of the construction geometry or two buttons, one to hide all constraints and one to hide all construction/internal geometry?
When hiding construction, I would also include to hide reference external geometry (not defining external).
Any feedback?

pierreporte, 2025-03-02:
Why would someone need to hide everything except defining geometry? It seems to me that it helps to visualize the contour that would be produced. In this case, #11482 would do exactly this, at least when the contour is closed. If the user wants to see open contour, though, the fill color wouldn’t help. I think that it should be a temporary switch withing the sketch, otherwise users will definitely wonder where everything else went because they would forget they toggle the mode (this is guaranteed).

@FEA-eng could you explain the intended workflow?

FEA-eng, 2025-03-02:
The idea is simple - to quickly hide all reference geometries and constraints and thus switch between everything visible:

[image]

and only defining geometry and constraints visible:

[image]

In order to make the sketch clearer by hiding the reference layer.

maxwxyz, 2025-06-19:
@FEA-eng could you check if https://wiki.freecad.org/Sketcher_SwitchVirtualSpace already is sufficient for your issue? It hides all constraints but not geometry (defining and construction)

maxwxyz, 2025-06-19:
So just defining geometry, as if the sketch is closed then?

FEA-eng, 2025-06-19:
> So just defining geometry, as if the sketch is closed then?

And non-reference constraints.