FreeCAD/FreeCAD #13321: PartDesign: datum plane parallel and in the middle between two faces

FEA-eng, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be nice to have an option in datum plane creation to place it in the middle between to selected faces, parallel to them. Currently, you may get this by default:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-24:
Even worse if the faces are skewed. Then just rotating 90 deg around one axis is not sufficient.

[image]

maxwxyz, 2025-01-22:
@PaddleStroke this should already be possible with core datums, right?

PaddleStroke, 2025-01-23:
@maxwxyz this is not a core datum problem, this is a 'attachment sucks in freecad' problem.

Which is a large project.

Jose the guy who made the 'MbD' workbench has a much cleaner attachment dialog btw.

So I think the DWG needs to come up with a clean proposal for the attachment dialog (there was already some proposal for the sketch creation). Then someone needs to implement it.

FEA-eng, 2025-01-23:
> [@maxwxyz](https://github.com/maxwxyz) this is not a core datum problem, this is a 'attachment sucks in freecad' problem.

Exactly, the attachment dialog is outdated and far from being user-friendly. It needs major overhaul at some point. It should be "smarter" (choose modes better fitting the selection) and throw less errors.

I’m not sure if it’s a good idea there’s a single attachment dialog for all datums and other features. In other CAD software, each datum type has its own modes. But I guess it would be hard to change it at this point.

FEA-eng, 2025-10-01:
Another example of why we need this:

[image]

Here, it's also necessary to select the cylindrical face to make it work. This is mentioned in #24247