FreeCAD/FreeCAD #13323: Display of face normals

FEA-eng, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Face normals are sometimes very important (e.g. in FEM). Currently, there's no easy way to check them apart from a [macro](https://forum.freecad.org/viewtopic.php?t=49276) and some tricks with backlight color (requiring non-default settings) also mentioned in that linked forum thread. An option to display arrows or temporarily color the face sides according to their normals would be very helpful.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-06-04:
> is this regarding face normals on solids or FEM meshes?

It's about normals of surface geometries that can later be used for FEA with 2D elements and the normals of elements (determined by geometry normals) are needed for pressure load, contact and so on. Plus in plane stress/strain and axisymmetric analyses, the geometry on the XY plane must have a positive normal facing the screen.

luzpaz, 2026-09-20:
Hi @tetektoza, are you still working on this ? (or should this be unassigned ?)

FEA-eng, 2026-09-20:
> Hi [@tetektoza](https://github.com/tetektoza), are you still working on this ? (or should this be unassigned ?)

From what I know, he left FreeCAD development for now, so I guess you can unassign him.