FreeCAD/FreeCAD #13324: Draft: uniform scaling option for Scale and Clone

FEA-eng, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Draft Scale and Clone tools could significantly benefit from having an option to apply uniform scaling (specify a single scale factor). Currently, it's necessary to enter the scale factors one by one and wait for the model to rebuild 3 times (which may take a while in the case of larger models). I know that a workaround could be to use Part Scale (which has the uniform scaling option) but it would still be nice to have uniform scaling available in Draft.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-06:
We need to distinguish the tool and the object here. The issue is about the Draft_Clone object. The Draft_Scale tool does have a uniform scale option (a checkbox).

There is no consistent way to handle scaling properties across FreeCAD:
* Draft_Clone uses 1 vector.
* Part_Scale has 4 inputs, 1 for uniform scaling and 3 for non-uniform.
* Link has 2 inputs, 1 for uniform scale and a vector for non-uniform scaling (you have to check "Show all" in the tree for the vector input).

One thing that might be implemented is to open a scaling dialog if a Draft_Clone is double-clicked in the Tree view.

FEA-eng, 2024-04-08:
> One thing that might be implemented is to open a scaling dialog if a Draft_Clone is double-clicked in the Tree view.

Right, that would be nice. I updated the issue following your feedback.

yorikvanhavre, 2024-04-16:
I would do the same as App::Link I guess... A  visible, uniform scaling prop and a hidden vector one for the few who need differential scaling. The problem is what updates what between these two... Should the uniform scaling "reset" the vector scaling prop when changed?

yorikvanhavre, 2024-04-22:
But then you have the risk to see your vector scaling values accidentally reset (for ex when loading a file). Maybe one way to solve this could be to say "if you want to use vector scaling, you must set uniform scaling to zero"

FEA-eng, 2024-04-22:
Or maybe the vector scaling fields could be disabled until a checkbox or true/false preference "Use non-uniform scaling" is enabled. Then the uniform scale field would become grayed out.

PR #19477 (2025-02-10): Draft: Introduce Edit option for Draft Clone scaling
Fixes #13324.

The Edit option is available from the Tree view context menu, but you can also double-click the clone in the Tree view.

The task panel is based on that of the Draft Scale command. The "Uniform scaling" checkbox is (un)checked depending on the current scaling of the clone.