FreeCAD/FreeCAD #13325: TechDraw: ActiveView should not be a tool, and shouldn't insert a png.

PaddleStroke, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, ActiveView tool insert a screenshot of the selected object as a view.

Ideally we would not need a tool. Instead we would have a way to select the 'Draw Style' of the view. And one of the options would be 'rendered'. See for example how another software handles this : 
[image]

Having said that I don't really know how we could achieve this. Because currently we project the shapes flat. So even if we could apply the color of the original faces (which we currently can't as we have no way to go from 2D to origin 3D), it would be without lightning effect. So it would look bad.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mpfusion, 2024-04-05:
A major benefit would be that the 3D render could update when the model is changed. Right now one has to delete the image and re-insert a new version once the model has been modified, which is easily forgotten and leads to the 3D render not reflecting the actual model.

WandererFan, 2024-04-06:
I don't know how to put the coin image into a page.   Maybe it is possible to use a Quarter widget as QGraphicsProxyWidget inside the scene?   That would solve the png part of the question, so the graphic would look better.

Not sure how to paint only selected objects on the Quarter widget without messing up the real 3d view.