FreeCAD/FreeCAD #13328: Non-streamlined context menu Property View

pieterhijma, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The context menu for properties in the Property View is not very streamlined for common actions.  In my opinion the most common action one would expect for a right-click in the property view is to add a property and remove a property.  In that sense it is related to #13038.  Instead of that, it shows the following three options that are not very useful:

[image]

Auto-expand is very user-specific I would think.  The value is remembered for the session so I would propose to move that to the second level that becomes visible by checking "Show all".  Expression is also not very useful because for the highlighted property it is also accessible by clicking the formula sign in the value input box, so it is superfluous.

In contrast, adding or removing a property is hidden behind the "Show all" flag and I would not be surprised if many users don't even know about this way of adding or removing properties.  In addition, both the Property View and the context menu itself become more complex because many more properties and actions are shown. This extra complexity in the Property View and context menu is not necessary to add or remove properties. The current second-level context menu:

[image]

I would suggest to make a small change to the context menu to streamline it for the most common usage.  In level 1 of the context menu we have:

Add Property
Remove Property (if applicable)
Show all

In level 2 if "Show all" is checked, we would have:

The ones above +
Auto expand
Expression... (I would be totally ok to remove this one altoghether)
Rename property group
Hidden
Output
NoRecompute
ReadOnly
Transient
Touched
EvalOnRestore
CopyOnChange

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36617 (Git)
Build type: Release
Branch: main
Hash: 4845d391ed2702b0b51a9da905066485eb5ef7bc
Python 3.11.6, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * backup 0.2.3
  * BIM 2021.12.0
  * Assembly3 0.12.2
  * Manipulator 1.5.7
  * fasteners 0.5.10
  * Assembly4 0.50.7
  * ondsel-lens 2024.3.26.01
  * DynamicData 2.60.0
```

### Subproject(s) affected?

Core

### Anything else?

This issue is related to #13038 and a bit related #12024.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-04-16:
I would suggest even more: Let's add some cool UI stuff to the property editor to do that. For ex:

* A "+" button somewhere, for ex here:

[image]

* That button opens the "add dynamic property" dialog

* That dialog maybe needs some rework, some options are IMHO a but unclear for new users

* Dynamic (removable)  properties should be somehow marked in the properties view, and indeed an easy way, but less obvious than a "-" button, to remove them (right-click?) offered. 

* But there are dynamic properties added by workbenches, which should not be removed by the user (although they in theory are), and user-added properties, which should be easy to remove. How to differentiate them?

pieterhijma, 2024-04-27:
> I would suggest even more: Let's add some cool UI stuff to the property editor to do that. For ex:
> 
> * A "+" button somewhere, for ex here:
> 
> ![Untitled](https://private-user-images.githubusercontent.com/1136856/322776991-08f40310-3719-49b1-87aa-7647feb24aa9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyMjA4NDAsIm5iZiI6MTcxNDIyMDU0MCwicGF0aCI6Ii8xMTM2ODU2LzMyMjc3Njk5MS0wOGY0MDMxMC0zNzE5LTQ5YjEtODdhYS03NjQ3ZmViMjRhYTkuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjdUMTIyMjIwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDk4MTc0NGFhZGJhZTlhMThmYzNhZWFlOTc4NWQ4Zjc4MjA0MDZkYjQzMTFkZDA1NzNkNWFjODBhOTNlMGQ3NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ZZqKpDkkge2fzixST6HRCB_g4YsVcZlOppo02VSXEas)

Cool idea, I'm not sure whether that is possible/wanted because of the fact that that space is reserved for tabs.

pieterhijma, 2024-04-27:
> * That dialog maybe needs some rework, some options are IMHO a but unclear for new users

Yes, I agree.

> * Dynamic (removable)  properties should be somehow marked in the properties view, and indeed an easy way, but less obvious than a "-" button, to remove them (right-click?) offered.

Yes, I think a right-click makes sense.
 
> * But there are dynamic properties added by workbenches, which should not be removed by the user (although they in theory are), and user-added properties, which should be easy to remove. How to differentiate them?

Hm, good question. @yorikvanhavre, do you have a concrete example of a dynamic property added by a workbench that shouldn't be removed? @FlachyJoe figured out something with property status [in this comment](https://github.com/FreeCAD/FreeCAD/issues/13139#issuecomment-2017907776). 

By the way, I in my humble opinion, the name "dynamic" for properties is not so useful for users. Doesn't it simply come from the fact that dynamic properties are added to objects at run-time, meaning that statically (in the code, at compile-time) no information is available about them. If this is the real differentiator between dynamic properties and non-dynamic properties, then I think we shouldn't bother users with this term. I think we don't even have to make a distinction in that case. For a user it is enough that some properties are hardcoded and cannot be removed and some can.

FlachyJoe, 2024-04-27:
@pieterhijma unfortunately `getPropertyStatus` returns `21` for **all** dynamically added properties, user and python ones.

[My user property list](https://forum.freecad.org/viewtopic.php?t=86378) displays FEM objects :cry: 
[image]

pieterhijma, 2024-04-27:
Right, that is a pity...

FlachyJoe, 2024-04-27:
Found the `LockDynamic` status to prevent the dynamically added property to be removed.
I'm making a PR to add this status to FEM object's.

FlachyJoe, 2024-04-27:
> Expression... (I would be totally ok to remove this one altoghether)

Not OK, it's the only way to set an expression to non-numeric properties.

pieterhijma, 2024-04-28:
> Found the `LockDynamic` status to prevent the dynamically added property to be removed. I'm making a PR to add this status to FEM object's. EDIT: #13687

Great! @yorikvanhavre and @FlachyJoe, what do you think? Should it become a convention for workbenches to distinct the properties that they deem important to mark them as such?

pieterhijma, 2024-04-28:
> > Expression... (I would be totally ok to remove this one altoghether)
> 
> Not OK, it's the only way to set an expression to non-numeric properties.

Yes, you're right. I figured that out myself as well. I never used it, so if you want to use an expression to set a boolean for example, you have to use this. It's a bit of a pity that the expression system only supports numeric values and strings.

pieterhijma, 2024-04-28:
I've submitted draft PR #13691 that rearranges the menu, keeping the "Expression..." item. For me, this would totally suffice, but an add button that @yorikvanhavre suggests, would be nice too.

pieterhijma, 2024-05-24:
So, a first step has merged by means of #13691. However, it would be good if the @FreeCAD/design-working-group could weigh in on this overall issue as well.

Based on [this comment](https://github.com/FreeCAD/FreeCAD/pull/13626#issuecomment-2129083759) I think it would be good to discuss adding GUI support for tooltips for properties as well.