FreeCAD/FreeCAD #13329: TechDraw: has 2 hatch face tools. One would do

PaddleStroke, 2024-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Techdraw has 2 hatch face tools when one tool would do perfectly the job.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-08:
Right, this duplication is confusing. There should be one tool with 2 ways of hatching - using svg or bitmap file or using predefined PAT patterns.

PaddleStroke, 2024-05-15:
@chennes is it possible to set up a grant/bounty on specific issue without doing it myself ? If yes what is the process to attach a bounty on an issue?

This project is middle size, I think a $500 bounty would be enough to get someone to jump in.

chennes, 2024-05-18:
We have "FPA designed grants" where we offer a fixed sum for a specific project. So if you want to write up a proposal we can discuss it at the next FPA meeting.