FreeCAD/FreeCAD #13336: Pyside2 and base::DIR_APP_DICTIONARIES error

sdifru, 2024-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I open FreeCAD, I get these two errors on report view:

1. `PySide2/__init__.py: Unable to import shiboken2 from <<followed by many directories to workbenches>>`. 
2. `Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/freecad/bin/qtwebengine_dictionaries'`2

### Full version info

```shell
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Python 3.11.8, Qt 6.6.3, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * PieMenu 1.4.2
  * Assembly3 0.12.2
  * fasteners 0.5.16
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-06:
is it also with the weekly builds? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

sdifru, 2024-04-06:
@maxwxyz I don't know whether or not it is a weekly build. On arch linux, freecad is updated every time there is a new release. Here is the link [freecad on arch linux](https://archlinux.org/packages/extra/x86_64/freecad/).

maxwxyz, 2024-04-06:
@sdifru no it't the current stable version: Version: 0.21.2.33771 (Git)
You can download a weekly dev version from here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

sdifru, 2024-04-06:
@maxwxyz Ah, I'm sorry I misunderstood your reply. I'll give the weekly build a try.

sdifru, 2024-04-07:
I found this error `PySide2/__init__.py: Unable to import shiboken2 from <<followed by many directories to workbenches>>` existed because I installed [this](https://github.com/Grubuntu/PieMenu) add-on.

This error `Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/freecad/bin/qtwebengine_dictionaries'` (or warning?) still persist.

I haven't tried the weekly build because I plan to use the stable version.

luzpaz, 2024-08-18:
Any Arch users able to reproduce on dev version ?

bdreddy, 2024-09-04:
I encountered this issue on Ubuntu 22 today. Does anyone have more information on the cause or solution?

luzpaz, 2024-09-05:
@bdreddy full [about](https://wiki.freecad.org/About) info please, thx

hasecilu, 2024-11-04:
It appeared to me when switching to Rocket WB but seems to work fine. Self-compiled Qt6.

Ui/TaskPanelStage.py
```python
try:
    from PySide2.QtWidgets import QDialog, QVBoxLayout
except:
    from PySide.QtWidgets import QDialog, QVBoxLayout
```
Another instance of PySide2 is on crowdin utility script, probably not longer needed? @davesrocketshop 

I don't see a task panel as in bosy, nose, transition, etc

```
OS: Arch Linux (Hyprland/hyprland/wayland)
Architecture: x86_64
Version: 1.1.0dev.36460 +2653 (Git)
Build type: Release
Branch: test/prs
Hash: 440b96eaf3a2a12a9079b3944101a11cb53a25fe
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Rocket 4.0.0
```

davesrocketshop, 2024-11-05:
This a problem with the Rocket WB and not FreeCAD. I thought I'd caught all thePySide2 references but apparently not. I've just published an update.

The references in the translation section won't go away soon, but that's not user executed code.

This issue can be closed.

davesrocketshop, 2024-11-05:
Correction: This has been fixed in the Rocket WB. I haven't (and won't) fix others.

This needs to be reported to the WB maintainers. This is not an issue with the core product.

luzpaz, 2024-11-06:
> This needs to be reported to the WB maintainers. This is not an issue with the core product.

@davesrocketshop can you please open a ticket on https://github.com/FreeCAD/FreeCAD-addons/issues related to this ?

hasecilu, 2024-11-06:
According to grep SteelColumn, Maker WOrkbench and freecad_metal_workbench are other WB that use PySide2, I don't know if replacing it with PySide is enough, not Qt guru.