FreeCAD/FreeCAD #13338: Hashing clip path is missing from svg export from the TechDraw workbench

dov, 2024-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

What I did:

1. Create new empty document
2. Switch to the Sketcher workbench, create sketch in the xy plane and add a rounded rectangle to it. Close sketcher
3. Switch to top view (2)
4. Switch to the TechDraw workbench
5. Add default page
6. Select the Sketch in the model tree and press Insert View
7. Select the rounded rectangle in TechDraw view. Then press "Hatch a file using image file". Change color to black and press ok.
8. Choose view tab and change the Face color to white (for better visibility)
9. Choose export svg 
10. Open the svg in an internal viewer

What I got:

Here is the resulting svg file, where the hashing flows outside the chosen face:

[image]

What I expected:

The hashing should be clipped to the face like in the TechDraw view:

[image]

I'm also attaching the resulting FCstd file in a zip file:

[bad-hashing-export.zip](https://github.com/FreeCAD/FreeCAD/files/14894739/bad-hashing-export.zip)

### Full version info

```shell
OS: Fedora Linux 39 (Workstation Edition) (fvwm)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36442 (Git)
Build type: Unknown
Branch: main
Hash: bcf54d3153fe4cad841ff7c0b692506eaa802103
Python 3.12.2, Qt 6.6.2, Coin 4.0.2, Vtk 9.2.6, OCC 7.8.1.dev
Locale: C/Default (C)
Installed mods: 
  * Curves 0.6.21
  * MyWorkbench
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dov, 2024-04-06:
I'm attaching here an example svg file created with inkscape illustrating how clipping can be done with svg.

[svg-clip-example.zip](https://github.com/FreeCAD/FreeCAD/files/14894769/svg-clip-example.zip)

maxwxyz, 2024-04-06:
Does https://wiki.freecad.org/TechDraw_GeometricHatch function produce the same outcome?

dov, 2024-04-06:
> Does https://wiki.freecad.org/TechDraw_GeometricHatch function produce the same outcome?

No, I just tested it. It does not! It's svg output renders ok.

[image]

Note however, that the neither svg, through Image File or GeometricHatch, makes use of the svg clipPath feature.

(Alltogether the svg output is extremely redundant, with e.g. setting the font-family element for every path element drawn)

maxwxyz, 2024-04-06:
@WandererFan FYI

WandererFan, 2024-04-08:
QtSvg only supports Svg-Tiny.  According to Adobe Illustrator Help -- Saving artwork: Save in SVG format, SVG Tiny does not support gradients, transparency, clipping, masks, symbols, patterns, underline text, strike through text, vertical text, or SVG filter effects.

This works in v0.20.2 but not in v0.21.2 or v0.22dev, so it is an old regression.

dov, 2024-04-12:
Yikes. Pixmaps for svg is ugly. :-(

How hard would it be to parse (a subset of) svg in the SVG tiles and then use libclipper for the clipping? E.g. using the first path statement only and then transform and clip by libclipper?

WandererFan, 2024-05-18:
What is libclipper?  I found this https://packages.debian.org/sid/libclipper-dev which doesn't sound like what we want.

Also this https://angusj.com/clipper2/Docs/Overview.htm, but it only works on straightline segments.

WandererFan, 2024-05-18:
I know how to parse Svg path, rect and line elements into QPainterPaths, but the paths don't get clipped on export.

Might need to convert the Svg into OCC edges and then use a boolean intersection with the face to be hatched.  This is how the PAT hatches work.

WandererFan, 2024-05-19:
related issue: Hatch not correctly exported to PDF #14027

WandererFan, 2024-10-19:
This is fixed:
[image]
[image]

closed 2024-10-19 by WandererFan