FreeCAD/FreeCAD #13340: Material: No actual preview when an object is selected

maxwxyz, 2024-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting an object and starting the material task dialog to apply a material, there is no live feedback of the actual appearance. This is because the entire object is highlighted with the selection color.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * kicadStepUpMod 11.1.0
  * lattice2 1.0.0
  * OpenDark 2023.12.17
  * Rocket 3.3.0
  * sheetmetal 0.4.10
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-04-06:
This is consistent with existing (pre-v0.22) behaviour. Are you suggesting that we not show the selection colour while the dialog is open?

maxwxyz, 2024-04-06:
Maybe related: In https://github.com/FreeCAD/FreeCAD/issues/13344 the body material is only visible when the selection color is overlayed

Cinnabar48, 2024-09-14:
the full shape highlight could be replaced with an outline like the blender "contour" or "external contour"
[image]