FreeCAD/FreeCAD #13341: Material: Changing a material is only visible after manual recompute

maxwxyz, 2024-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When clicking on a feature of a model, e.g. an edge and starting the material dialog there is only a subtle change when switching through the different materials. To see the real material I have to manually recompute after each change.

https://github.com/FreeCAD/FreeCAD/assets/6246609/76746c94-2fdd-415f-ac8d-fd89cb7d01a1

Why do I only select an edge to begin with: https://github.com/FreeCAD/FreeCAD/issues/13340

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * kicadStepUpMod 11.1.0
  * lattice2 1.0.0
  * OpenDark 2023.12.17
  * Rocket 3.3.0
  * sheetmetal 0.4.10
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

berberic2, 2024-04-09:
Same problem here with 

```
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36790 +2 (Git)
Build type: Release
```

BTW: You can change `Selection Style` of the object to `BoundBox` to be able to select the object and see colour changes immediately.

wwmayer, 2024-04-20:
Is this still reproducible with the latest weekly version?

For me it works correctly:
```
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36905 (Git)
Build type: debug
Branch: main
Hash: 1a2e922dce6167faca6c69f660d9e48521fb6a01
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

maxwxyz, 2024-04-20:
@wwmayer thanks, its working for me with
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.34
  * fasteners 0.5.17
  * freecad.gears 1.2.0
  * OpenDark 2024.3.13
  * sheetmetal 0.4.11

```

closed 2024-04-20 by maxwxyz

berberic2, 2024-04-20:
> Is this still reproducible with the latest weekly version?

No. Newly build from git (0.22.0dev.36911 +2) and works fine here now.