FreeCAD/FreeCAD #13343: Material: properties are saved when clicking cancel

maxwxyz, 2024-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When editing properties with matrix input, e.g. Yeoh, the values in the table are saved, even when clicking cancel.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * kicadStepUpMod 11.1.0
  * lattice2 1.0.0
  * OpenDark 2023.12.17
  * Rocket 3.3.0
  * sheetmetal 0.4.10
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-08-15:
The data is being updated "live" as you type, so neither OK nor Cancel actually do anything at all. The two paths forward are to not update the data until OK is clicked, or to save a copy of the original when the dialog shows, and make cancel restore it when it is clicked.

luzpaz, 2024-10-26:
Any chance this gets addressed in 1.0 ?

davesrocketshop, 2024-10-26:
> Any chance this gets addressed in 1.0 ?

Given that no WB uses this at the moment it's pretty low priority for 1.0

maxwxyz, 2025-09-25:
Still relevant

FEA-eng, 2026-09-30:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/49a222e06017d004042f6a6a7a1a30f902101105

The 2D array editor keeps changes private until OK, so Cancel leaves the material unchanged. The separate 3D array editor is not covered.