FreeCAD/FreeCAD #13346: einzelne Vorschaubilder aus Startseite entfernen

HKaufmann2, 2024-04-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Wenn man viel experimentiert entstehen viele Vorschaubilder die von den wichtigen Dateien ablenken.
Deswegen wäre es gut wenn man die unwichtige Vorschaubilder entfernen könnten. z.B. über die rechte Maustaste => entfernen. Es sollen **nicht** die Daten gelöscht werden.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-07:
Please create the issue in English. Which thumbnail images do you mean, can you provide a screenshot?
Bitte probiere mal die letzten Weekly Builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Die Startseite wurde dort bereits überarbeitet, ist aber noch im Prozess.

HKaufmann2, 2024-04-07:
Hallo maxwxyz,
kann kein Englisch.
[image]

chrisb-github, 2024-04-07:
It's a feature request following this (German) discussion: https://forum.freecad.org/viewtopic.php?t=86729 . 
HKaufmann wants to remove the thumbnails of _some_ files.
Currently the thumbnail creation is controlled in the preferences, so that either all or none is created with a thumbnail. To implement this, it may need an additional property for the document.
@HKaufmann2, feel free to reopen the issue.

FEA-eng, 2024-04-08:
Yes, there should be a way to remove selected recent file thumbnails:

[image]

maxwxyz, 2024-04-08:
I think based on the forum discussion it is meant to delete the item from the recent file list, not just the image.

FEA-eng, 2024-04-08:
> I think based on the forum discussion it is meant to delete the item from the recent file list, not just the image.

Yes, I'm thinking about deleting the whole entry, not just the image. In my example, if I click Remove in SW, the whole tile (image, name and frame) will disappear making place for another one. Of course, the file itself won't be removed from the computer:

> Therefore, it would be good if you could remove the unimportant thumbnails. by using the right mouse button >. **The data should not be deleted.**

maxwxyz, 2024-04-08:
@HKaufmann2 bitte um kurze Bestätigung: Es geht dir darum, die einzelnen Einträge aus der Liste der letzten Dateien zu löschen, ohne dabei die Datei selbst zu löschen, richtig? Es geht dir nicht darum, nur die Vorschaubilder zu löschen, und stattdessen die selbe Datei ohne ein Bild in der Liste anzuzeigen.

HKaufmann2, 2024-04-09:
Hallo maxwxyz, es sollen nur die Vorschaubilder aus der Liste entfernbar sein.
Hello maxwxyz, only the thumbnails should be removable from the list.

jorgebr, 2024-08-30:
For simple users just searching for a way how to do this, even if it's not part of the user interface:
Go to Tools > Edit Parameters ...
Then BaseApp > Preferences > RecentFiles

You can edit or delete any entries there. You must restart FreeCAD to see the changes.