FreeCAD/FreeCAD #13347: TechDraw: ProjectShape tool is in wrong workbench. Should be in Draft.

PaddleStroke, 2024-04-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[TechDraw_ProjectShape](https://forum.freecad.org/viewtopic.php?t=86719) does not belong to techdraw workbench. It should be moved to Draft (or Part alternatively).
@yorikvanhavre Are you interested in moving this tool to Draft?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-07:
Right, it shouldn't belong to TD. It should probably go to Draft like Shape2DView but Part would also make sense.

luzpaz, 2024-04-07:
cc @Roy-043

PaddleStroke, 2024-04-10:
Hmm this Shape2DView seems very similar to ProjectShape. But I don't know these tools well so I can't say for sure. Perhaps it is not worth keeping both then? Should be just remove ProjectShape? @yorikvanhavre ?

yorikvanhavre, 2024-04-16:
Draft_Shape2DView has a few additional functionalities such as the ability to fill closed areas or show cut lines. OTOH TD projectShape is probably faster. So at the moment although they are similar I can't say one should be removed.

Having both in Draft is a possible, but IMHO this will not make things easier to the user... There will now be two very similar tools one next to the other.

Having it in Part wouldn't be good because it would make Part depend on TechDraw when it's currently the contrary.

What about keeping it in TD but putting in some kind of "utils" submenu?

PaddleStroke, 2024-04-16:
If `Draft_Shape2DView `is a super set of `TechDraw_ProjectShape`, then there is no reason to keep it around. Unless it is indeed significantly faster. Could you please make a little test to see if it is actually faster?

- If it is not faster then we can just remove TechDraw_ProjectShape.
- If it is faster, is there not a way to make `Draft_Shape2DView` faster by using what makes `TechDraw_ProjectShape` faster?
- If it is faster but speed cannot be transfered to Draft_Shape2DView, then having both in draft together would make most sense imo, in a command group for example. With tooltip that explain the differences.

If we move to Part, we don't have to make Part depend on TD, we can move the code to Part probably not just the tool.

Leave in TD but only in menu, I think this is not as good as a solution. It will just be completely hidden and cluttering a bit.