FreeCAD/FreeCAD #13349: Transparent Overlays: Panel names are always visible, when hovering twice

maxwxyz, 2024-04-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The title names of the panels / docked windows are always visible when using the transparent overlay mode:
[image]

When hovering the panels, they are even displayed twice. I'm not sure when this was added or why but a few weeks ago there were no default titles displayed as seen in a screenshot from https://github.com/FreeCAD/FreeCAD/issues/11794
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * kicadStepUpMod 11.1.0
  * lattice2 1.0.0
  * OpenDark 2023.12.17
  * Rocket 3.3.0
  * sheetmetal 0.4.10
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jag45, 2024-04-07:
This coincided with the following change:
[image]

maxwxyz, 2024-04-07:
@LemonBoy can you confirm?

PaddleStroke, 2024-04-08:
+1 I can reproduce.
- Now we have 2 bars. 
- I agree the situation was better before. I do not see value of having the title bars displayed all the time. It was better when it shows only on hover.
- The new bar looks better imo. But it lacks the settings button. It also lacks the resize handle.

[image]
Btw it would be better to remove the eye icon and the overlay icon and put these options in the 'settings' button.

jag45, 2024-04-08:
Tried with weekly 36228, only the one bar when you hover.
Tried with weekly 36277 (released 6th March), we have the situation above, ie. 2 bars.

As an aside , I cannot find lemonboy in the member list. However there is a lemonbug, (pretty close on the keyboard) so maybe the WIP Wednesday screenshot above has a typo on the user name.

maxwxyz, 2024-04-19:
@ayushgupta9906 any news on this?
@LemonBoy could you weigh in, was it introduced with your commit?

MisterMakerNL, 2024-04-20:
I just build the latest had similar issue added a gif where is visible here:
https://github.com/FreeCAD/FreeCAD/pull/12573 (which is probably related since it is also overlay UI stuff.)

maxwxyz, 2024-04-20:
@kadet1090 could you support here?

maxwxyz, 2024-04-20:
is https://github.com/FreeCAD/FreeCAD/pull/12530 to blame?

MisterMakerNL, 2024-04-20:
Double title seems to go away when you undock the widget, so maybe it has to do with:
`OverlayManager::onDockVisibleChange`
Or maybe it can be fixed with that function?

kadet1090, 2024-04-21:
> @kadet1090 could you support here?

I'll look into that this week hopefully

maxwxyz, 2024-04-29:
@kadet1090 do you need help with testing or anything else I could support?

kadet1090, 2024-05-05:
I fixed it in #13838, but only the issue with double header. Any remodeling of the header will need to be done in https://github.com/FreeCAD/FreeCAD/pull/11998 as current implementation is tape over tape...

maxwxyz, 2024-05-05:
nice, thanks!

MisterMakerNL, 2024-05-05:
Finally the double header it's gone! Tnxs!!!

closed 2024-05-06 by yorikvanhavre

MisterMakerNL, 2024-05-06:
Well it is fixed a bit sometimes :)
Needs some work!

PR #13838 (2024-05-06): Gui: Remove redundant overlay header
Problem was caused by replacing the widget even if it should be invisible. 

Fixes: #13349
commit 075f3872 (2024-05-06): Gui: Remove redundant overlay header

Fixes: #13349