FreeCAD/FreeCAD #13350: Transparent Overlays: Property View cells allow click through

maxwxyz, 2024-04-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The click through in the tree view is good in general but it becomes problematic for the property view with transparent overlays enabled. It is not possible to select input fields which have no or little text in it, as they only work when you directly click on the test. If you click in the same cell a bit too far to the right, the object is deselected and the property view is empty again.
Especially with the input fields which have `...` on the right side of the cell (e.g. placement) it is not possible to select or press this button. You have to first select the cell on the left most part, then the `...`becomes visible and you can click it. Same with dropdown or expressions in the property view.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/eaf644e7-ea96-4de5-9756-ae1f0b8c9a9f

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 (Git)
Build type: Release
Branch: main
Hash: 6ca35709ddc7f95fb84eef4c24973dc489e5acde
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * kicadStepUpMod 11.1.0
  * lattice2 1.0.0
  * OpenDark 2023.12.17
  * Rocket 3.3.0
  * sheetmetal 0.4.10
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-07:
@Rexbas FYI, you just fixed the click through issue with navigation

maxwxyz, 2024-08-25:
this is obsolete as the cells are not transparent anymore thanks to @kadet1090

closed 2024-08-25 by maxwxyz

luzpaz, 2024-08-25:
@maxwxyz what PR or commit is the fix related to ?

maxwxyz, 2024-08-25:
https://github.com/FreeCAD/FreeCAD/pull/15328