FreeCAD/FreeCAD #13351: Transparent Overlays: Property View is not transparent in some themes

Rexbas, 2024-04-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Dark, Dark Modern, Darker, Light and Light Modern theme, the Property View is not transparent with transparent overlays enabled.
[image]

### Full version info

```shell
OS: KDE neon 6.0 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36739 (Git)
Build type: Release
Branch: main
Hash: b4e50f4d9347260dc9f05e39936a77cbbf5ffc96
Python 3.11.6, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-07:
@MisterMakerNL FYI

MisterMakerNL, 2024-04-07:
This is on purpose to work around a bug. I am not able to set the background to be only on when the properties panel is shown. 
It's currently this or no background on certain cells.
And since no-background would make some text unreadable it ended to be this.

maxwxyz, 2024-05-02:
@MisterMakerNL like this bug? https://github.com/obelisk79/OpenDark/issues/67

MisterMakerNL, 2024-05-02:
Yes, it's annoying.
I can only set alternate cell background but not the other version.

maxwxyz, 2024-05-02:
But I'd rather have it not transparent then unreadable and not able to click on the entire cell. It's only in the property view anyways.

MisterMakerNL, 2024-05-02:
I think we have to dig into @realthunder code to try to fix this. 
The table:item has no background so nothing to change in QSS files except giving the complete table a background.
I read the code, have no idea what I read ;)

MisterMakerNL, 2024-05-11:
I can fix the true click and make it a bit more pretty by giving the pane a 0.01 transparency:
[image]
[image]

luzpaz, 2024-07-27:
What's left to do here ?

maxwxyz, 2024-07-27:
It is fixed in the latest weekly.
@Rexbas if there is still an issue, please reopen.

closed 2024-07-27 by maxwxyz