FreeCAD/FreeCAD #13355: Strange anomaly after polar-then-groove

Malendryn, 2024-04-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a gear I'm designing, some (but not all!) of the points of the final part are improperly computed.  Just spitballing, but it seems to somehow be related to the polar pattern processing?

I've attached a .fcSTD file that's pared down to the absolute minimum part

### Full version info

```shell
[code]

--  NOTE:  Designing in release build, but ALSO happening in latest 0.22.0 devel build  --

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.15
  * freecad.gears 1.2.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]
[Anomaly.zip](https://github.com/FreeCAD/FreeCAD/files/14898531/Anomaly.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-07:
Try again with Refine set to true for all features - you can even set this in preferences:

[image]

FlachyJoe, 2024-04-08:
Without surprise it fails with 0.22.0dev too. It's an OCC failure.

It works if you groove the tooth before the polar.
[image]

Malendryn, 2024-04-08:
I tried moving the polar after the groove, but for me it caused the model to disappear completely. (Entirely possible I needed to click on the polar object to make it visible,  wasn't paying that close attention at the time :)

But what I did do, is just create the part without polar, then did the polar rot in parts (not design) workbench, and it worked fine when I did it that way.