FreeCAD/FreeCAD #13357: Crash on changing headlight direction

chrisb-github, 2024-04-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the current defaults sketcher is unusable out of the box, if there is an object in the background, see https://forum.freecad.org/viewtopic.php?p=752291#p752291 .
Changing the headlight direction should be a workaround, yet it crashes FreeCAD.

To reproduce:
- Start FreeCAD with a plain vanilla config
- Preferences->Display->Light sources
- Change the direction, e.g. to this:
[image]
- Click Ok
-> FreeCAD terminates without further notice.

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.5.15
  * ExplodedAssembly
  * Curves 0.6.29
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-08:
@wwmayer FYI

@chrisb-github usability issue: https://github.com/FreeCAD/FreeCAD/issues/11166

chrisb-github, 2024-04-08:
FreeCAD crashes as well if I cancel the preferences dialog after changing the headlight direction.

FEA-eng, 2024-04-08:
Can it be OS-specific? It doesn't crash for me when using non-vanilla config on Windows.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

xtemp09, 2024-04-09:
No crash here.
```ruby

OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Debug
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)

```

luzpaz, 2024-04-14:
cc @JohnOCFII can you reproduce on macOS ? TIA

JohnOCFII, 2024-04-14:
> cc @JohnOCFII can you reproduce on macOS ? TIA

Yes, it crashes for me every time as well.  

```
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

Crash Report:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [25408]
Path:                  /Applications/Other Apps/FreeCAD-36807.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [25402]
User ID:               501

Date/Time:             2024-04-14 12:48:54.2587 -0500
OS Version:            macOS 14.4.1 (23E224)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       E597DF5B-0146-41EC-81E1-266677F952EE

Time Awake Since Boot: 240000 seconds
Time Since Wake:       15666 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [25408]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4307632112
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100c14000-100c34000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-36807.app/Contents/Resources/bin/freecad

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libCoin.80.0.2.dylib          	       0x1053855a4 SoCallbackAction::~SoCallbackAction() + 124
1   libCoin.80.0.2.dylib          	       0x1053856d4 SoCallbackAction::~SoCallbackAction() + 12
2   libCoin.80.0.2.dylib          	       0x1053e2ea0 SoDragger::~SoDragger() + 192
3   libCoin.80.0.2.dylib          	       0x105404ff8 SoRotateSphericalDragger::~SoRotateSphericalDragger() + 12
4   libCoin.80.0.2.dylib          	       0x1055235a8 SoBase::destroy() + 212
5   libCoin.80.0.2.dylib          	       0x105517674 SoBaseList::truncate(int) + 80
6   libCoin.80.0.2.dylib          	       0x1055278c4 SoChildList::~SoChildList() + 24
7   libCoin.80.0.2.dylib          	       0x105573318 SoGroup::~SoGroup() + 40
8   libCoin.80.0.2.dylib          	       0x10558d600 SoSeparator::~SoSeparator() + 12
9   libCoin.80.0.2.dylib          	       0x1055235a8 SoBase::destroy() + 212
10  libCoin.80.0.2.dylib          	       0x105517674 SoBaseList::truncate(int) + 80
11  libCoin.80.0.2.dylib          	       0x1055278c4 SoChildList::~SoChildList() + 24
12  libCoin.80.0.2.dylib          	       0x10554fb10 SoBaseKit::~SoBaseKit() + 40
13  libCoin.80.0.2.dylib          	       0x1053ebd34 SoDirectionalLightDragger::~SoDirectionalLightDragger() + 12
14  libCoin.80.0.2.dylib          	       0x1055235a8 SoBase::destroy() + 212
15  libCoin.80.0.2.dylib          	       0x1055177d4 SoBaseList::remove(int) + 44
16  libCoin.80.0.2.dylib          	       0x105573884 SoGroup::removeChild(int) + 72
17  libFreeCADGui.dylib           	       0x1029c8fa0 Gui::coinRemoveAllChildren(SoGroup*) + 676
18  libFreeCADGui.dylib           	       0x1029936b4 Gui::View3DInventorViewer::~View3DInventorViewer() + 232
19  libFreeCADGui.dylib           	       0x102994020 Gui::View3DInventorViewer::~View3DInventorViewer() + 12
20  libFreeCADGui.dylib           	       0x1027d7620 Gui::Dialog::DlgSettingsLightSources::~DlgSettingsLightSources() + 56
21  libQt5Core.5.15.13.dylib      	       0x104925ad8 QObjectPrivate::deleteChildren() + 108
22  libFreeCADGui.dylib           	       0x1026736c4 Gui::Command::_invoke(int, bool) + 516
23  libFreeCADGui.dylib           	       0x102673484 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 348
24  libQt5Core.5.15.13.dylib      	       0x104929e84 0x1047e4000 + 1334916
25  libQt5Core.5.15.13.dylib      	       0x10490d094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
26  CoreFoundation                	       0x196d41e44 __CFRunLoopDoSource0 + 176
27  CoreFoundation                	       0x196d41bb4 __CFRunLoopDoSources0 + 244
28  CoreFoundation                	       0x196d407a0 __CFRunLoopRun + 828
29  CoreFoundation                	       0x196d3fe0c CFRunLoopRunSpecific + 608
30  HIToolbox                     	       0x1a14db000 RunCurrentEventLoopInMode + 292
31  HIToolbox                     	       0x1a14dac90 ReceiveNextEventCommon + 220
32  HIToolbox                     	       0x1a14dab94 _BlockUntilNextEventMatchingListInModeWithFilter + 76
33  AppKit                        	       0x19a598970 _DPSNextEvent + 660
34  AppKit                        	       0x19ad8adec -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
35  AppKit                        	       0x19a58bcb8 -[NSApplication run] + 476
36  libqcocoa.dylib               	       0x105c044d0 0x105bd8000 + 181456
37  freecad                       	       0x100c1b030 main + 4932
38  dyld                          	       0x1968da0e0 start + 2360

Thread 1:
0   libsystem_pthread.dylib       	       0x196c5dd20 start_wqthread + 0
```

wwmayer, 2024-04-14:
Where exactly do the binaries of the Coin3D library come from? And do you know whether patches are applied to the original source tarball?

I recently downloaded Coin-4.0.2 and made a FreeCAD build linking to that version but I didn't get this crash.

maxwxyz, 2024-06-15:
@JohnOCFII ping

JohnOCFII, 2024-06-15:
It still crashes for me on macOS:

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0

```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [6194]
Path:                  /Applications/Other Apps/FreeCAD-37709.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [6188]
User ID:               501

Date/Time:             2024-06-15 08:21:17.1043 -0500
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       D8658B1A-7473-4FE1-A036-5E92325C28C6

Time Awake Since Boot: 61000 seconds
Time Since Wake:       217 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [6194]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4339646448
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102a9c000-102abc000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-37709.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.2.dylib          	       0x1072b55a4 SoCallbackAction::~SoCallbackAction() + 124
1   libCoin.80.0.2.dylib          	       0x1072b56d4 SoCallbackAction::~SoCallbackAction() + 12
2   libCoin.80.0.2.dylib          	       0x107312ea0 SoDragger::~SoDragger() + 192
3   libCoin.80.0.2.dylib          	       0x107334ff8 SoRotateSphericalDragger::~SoRotateSphericalDragger() + 12
4   libCoin.80.0.2.dylib          	       0x1074535a8 SoBase::destroy() + 212
5   libCoin.80.0.2.dylib          	       0x107447674 SoBaseList::truncate(int) + 80
6   libCoin.80.0.2.dylib          	       0x1074578c4 SoChildList::~SoChildList() + 24
7   libCoin.80.0.2.dylib          	       0x1074a3318 SoGroup::~SoGroup() + 40
8   libCoin.80.0.2.dylib          	       0x1074bd600 SoSeparator::~SoSeparator() + 12
9   libCoin.80.0.2.dylib          	       0x1074535a8 SoBase::destroy() + 212
10  libCoin.80.0.2.dylib          	       0x107447674 SoBaseList::truncate(int) + 80
11  libCoin.80.0.2.dylib          	       0x1074578c4 SoChildList::~SoChildList() + 24
12  libCoin.80.0.2.dylib          	       0x10747fb10 SoBaseKit::~SoBaseKit() + 40
13  libCoin.80.0.2.dylib          	       0x10731bd34 SoDirectionalLightDragger::~SoDirectionalLightDragger() + 12
14  libCoin.80.0.2.dylib          	       0x1074535a8 SoBase::destroy() + 212
15  libCoin.80.0.2.dylib          	       0x1074477d4 SoBaseList::remove(int) + 44
16  libCoin.80.0.2.dylib          	       0x1074a3884 SoGroup::removeChild(int) + 72
17  libFreeCADGui.dylib           	       0x1048b7044 Gui::coinRemoveAllChildren(SoGroup*) + 676
18  libFreeCADGui.dylib           	       0x104880cd8 Gui::View3DInventorViewer::~View3DInventorViewer() + 232
19  libFreeCADGui.dylib           	       0x104881644 Gui::View3DInventorViewer::~View3DInventorViewer() + 12
20  libFreeCADGui.dylib           	       0x1046b9358 Gui::Dialog::DlgSettingsLightSources::~DlgSettingsLightSources() + 56
21  libQt5Core.5.15.13.dylib      	       0x106855ad8 QObjectPrivate::deleteChildren() + 108
22  libFreeCADGui.dylib           	       0x1045411a0 Gui::Command::_invoke(int, bool) + 516
23  libFreeCADGui.dylib           	       0x104540f60 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 348
24  libQt5Core.5.15.13.dylib      	       0x106859e84 0x106714000 + 1334916
25  libQt5Core.5.15.13.dylib      	       0x10683d094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
26  CoreFoundation                	       0x19d14646c __CFRunLoopDoSource0 + 176
27  CoreFoundation                	       0x19d1461dc __CFRunLoopDoSources0 + 244
28  CoreFoundation                	       0x19d144dc8 __CFRunLoopRun + 828
29  CoreFoundation                	       0x19d144434 CFRunLoopRunSpecific + 608
30  HIToolbox                     	       0x1a78e819c RunCurrentEventLoopInMode + 292
31  HIToolbox                     	       0x1a78e7e2c ReceiveNextEventCommon + 220
32  HIToolbox                     	       0x1a78e7d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
33  AppKit                        	       0x1a09a3d68 _DPSNextEvent + 660
34  AppKit                        	       0x1a1199808 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
35  AppKit                        	       0x1a099709c -[NSApplication run] + 476
36  libqcocoa.dylib               	       0x107b344d0 0x107b08000 + 181456
37  freecad                       	       0x102aa329c main + 3356
38  dyld                          	       0x19ccde0e0 start + 2360

Thread 1:
0   libsystem_pthread.dylib       	       0x19d061d20 start_wqthread + 0
```

maxwxyz, 2024-06-30:
can confirm it crashes on my mac also every time.

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * CfdOF 1.26.1
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.20
  * Curves 0.6.40

```

chennes, 2024-07-13:
With an older Coin, it does not crash:
```
OS: macOS 13.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38131 (Git)
Build type: Release
Branch: main
Hash: 9a51e2c6f1ae8b4a1919f6af0f90df6833ecf82f
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.5.3
Locale: English/United States (en_US)
```

maxwxyz, 2024-07-24:
Also, just opening this preference page and clicking ok without changing anything, changes the headlight direction. A reset does not reset it to (0,0,-1) properly.

On a fresh install it is 0,0,-1. When you just open the preferences and go to the Display>Light source tab the headlight direction changes to something to the top left side. When resetting the preferences it does not change.
It seems like the headlight is controlled by either HeadlightDirection or the vector of HeadlightRotationW,  X, Y, Z,... it seems the light adjusts to the parameter which was updated last, either direction or the rotation values. Not sure if this can be fixed for 1.0

maxwxyz, 2024-07-30:
On Windows with a Qt6 build the preference window instantly closes when navigating to this page Display > Light sources.
There is an access violation error.
@wwmayer FYI

chennes, 2024-07-31:
That's an odd one: the segfault there is happening in the notification system. The message it's trying to display is
```
"QLayout: Attempting to add QLayout "" to QGroupBox "groupBoxLight", which already has a layout\n"
```
That obviously shouldn't be a fatal error, but neither should writing to the console ever produce a segfault. And of course it's not clear if this is really what's causing the crash, or if it's just another symptom. We probably need to look at this with valgrind or the like.

xtemp09, 2024-07-31:
@maxwxyz, @chennes, could you both provide About info?

chennes, 2024-08-05:
I'm making some headway on this. First, the most obvious problem is that the DlgSettingsLightSources's destructor is deleting an unowned pointer to a QWidget, whose memory should be getting managed by Qt once its added to a layout. I suspect that line alone is causing the segfault that we're seeing on the Mac. Additionally, in `showEvent` there is a redundant call to `addLayout`: creating the layout with the widget as its parent has already done that, hence *one* of the warnings. But the second warning is more mysterious: if I replace the code in `showEvent` with
```
    if (!view) {
        QGroupBox* box = ui->groupBoxLight;
        QWidget* widget = createViewer(box);
        if (!box->layout()) {
            auto grid = new QGridLayout(box);
            grid->addWidget(widget);
        }
        else {
            box->layout()->addWidget(widget);
        }

        loadDirection();
    }
```
I actually end up with *two* copies of the view widget in the layout! But that should not be possible, `view` is only allocated once. Does anyone see the trick in there?

chennes, 2024-08-05:
Answer: it's recursing, the showEvent is getting called by itself -- so you always end up with two copies of the view widget.

commit 362cf1c8 (2024-08-09): Gui: Fix crash in DlgSettingsLightSources

Fixes #13357
closed 2024-08-09 by adrianinsaval