FreeCAD/FreeCAD #13359: Some shortcuts aren't rendered in Sketcher's geometry toolbar

prokoudine, 2024-04-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sketcher is inconsistent in displaying keyboard shortcuts for geometry tools in the toolbar.

In general, Sketcher:

1. Displays a shortcut in a tooltip when the toolbutton doesn't have a drop-down menu with tool variants

[image]

2. Doesn't display a shortcut in a tooltip when the toolbutton has a drop-down menu with tool variants

[image]

3. Mostly doesn't display shortcuts next to menu item titles in drop-down menus with tool variants...

[image]

4. ...with the sole exception of Slot, Fillet, and Trim.

[image]

I'm not sure if this is Sketcher-specific though.

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git) AppImage
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * DynamicData 2.60.0
  * Assembly4 0.50.6
  * ExplodedAssembly
  * sheetmetal 0.3.15
  * BIM 2021.12.0
  * A2plus 0.4.61
  * Render 2023.12.28
  * Rocket 3.2.4
  * Curves 0.6.21
  * freecad.gears 1.2.0
  * Assembly3 0.12.0
  * boltsfc 2022.11.5
  * OpenDark 2023.12.17
  * dodo 1.0.0
  * Ondsel-Lens 2023.12.13.01
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-08:
Would the tooltips overlay the nested commands?

prokoudine, 2024-04-08:
When you open the drop-down menu with nested commands, you don't get to see the tooltip for the parent comman anymore anyway.

Foadsf, 2024-08-02:
I started a new discussion [here](https://forum.freecad.org/viewtopic.php?t=89548) on the forum.

ruvilonix, 2024-11-04:
```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

I also experience this with 1.0 RC2. 

[image]
[image]
[image]

It would also be nice if the ones with a dropdown showed the shortcut even if just hovering over the default option and not expanding the dropdown. 
[image]
[image]

G0rocks, 2024-11-20:
There exist shortcuts for a bunch of the things that don't show up in the UI when you hover for example. I like that in sketcher when you hover Create line it shows (G, L) as the shortcut.
Here's the list:
https://wiki.freecad.org/Sandbox:Keyboard_Shortcuts

Kirmy34, 2026-11-08:
If nobody is on this I'd like to give it a go.

Foadsf, 2026-11-08:
I think, I have addressed many of these issues in 

https://github.com/Foadsf/freecad-nxstyle

already.