FreeCAD/FreeCAD #13360: FEM Analysis container should be separated from the rest of the tree

FEA-eng, 2024-04-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I don't know if it's doable without too much effort but the FEM Analysis container should be added separately to the tree so that it doesn't interfere with the geometry added after its creation:

[image]

In SolidWorks, the model and analysis trees are completely separated:

[image]

What's interesting, Netgen can mesh shapes below the Analysis container but Gmsh can't do it (throws `list index out of range` error).

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-08:
not sure about this, all features appear in the tree view. Otherwise you would have separate views/windows per WB.
You can now manually rearrange them in the tree. I think this needs more discussion.

FEA-eng, 2024-04-08:
> not sure about this, all features appear in the tree view. Otherwise you would have separate views/windows per WB. 

I'm not necessarily thinking about putting the trees in different windows but rather about making the FEM Analysis container always stay below the geometry one, if possible so maybe reordering by default?

> You can now manually rearrange them in the tree.

Reordering of the tree items is indeed possible but I would be careful with it and it may not work in all the cases.

pierreporte, 2024-04-08:
It would also be interesting if this container could be saved in a separate file. Analysis are heavy so it would save loading time. They rarely need to be that accessible since you’re either doing CAD or simulation, never both. It may be similar to #12708.

FEA-eng, 2024-04-08:
> Analysis are heavy so it would save loading time.

It shouldn’t be a problem if you clear the mesh (right-click and select Clear FEM mesh) and delete the results. The downside is that you lose the pipeline settings this way. Maybe there should be also an option to clear the results.

pierreporte, 2024-04-08:
The problem with clearing things is that you clear them. You can’t run a new simulation easily if you have to recreate the mesh (it may be fast for simple parts but what about large assemblies?). You can’t view the  results later if you delete them. If everything can be saved elsewhere, the problem is solved. It would also allow users to open only the simulation and not the CAD data.

FlachyJoe, 2024-04-09:
> It would also be interesting if this container could be saved in a separate file.

You can still use App::Link to separate geometry file(s) and analysis.

FEA-eng, 2024-04-09:
> it may not work in all cases.

It seems that manual reordering doesn't work when there's a single geometry object and it gets below the Analysis container:

[image]

maxwxyz, 2024-04-09:
@PaddleStroke ^ FYI

PaddleStroke, 2024-04-09:
I'm not sure but I would guess that this is specific to this object. Can you reorder other things correctly @FEA-eng ?
If its specific to this object it probable has something to do with the drag / drop functions of the view provider.

FEA-eng, 2024-04-09:
> Can you reorder other things correctly @FEA-eng ?

I can't reorder whenever there's a geometry object below an Analysis container. Another example:

[image]

The same happens with a Body container. Interestingly, I can drag the geometry object to put it below the Analysis container but then I can't move it back to the position above it.

marioalexis84, 2024-04-09:
>What's interesting, Netgen can mesh shapes below the Analysis container but Gmsh can't do it (throws list index out of range error). Maybe this should be a separate issue?

Yes. Please, create a separate issue.

FEA-eng, 2024-04-09:
> Yes. Please, create a separate issue.

Done: #13364

I'll create one for the drag-and-drop issue discussed with Paddle too since the idea of some separation or automatic sorting of geometry objects and Analysis containers in the tree is still valid.