FreeCAD/FreeCAD #13361: Bending existing parts

FEA-eng, 2024-04-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It has been requested a few times on the forum but I haven't found a GitHub issue for it. A tool to bend existing parts (other than sheet metal) could be useful but probably difficult to implement (I doubt OCC has any ready solution for that).

Example from SW:

[image]

A workaround, apart from using the [AddFoldWall](https://wiki.freecad.org/SheetMetal_AddFoldWall) tool in the add-on SheetMetal workbench, could be to bend the part in FEM and [export it to STL](https://wiki.freecad.org/FEM_FemMesh2Mesh) but of course meshes are harder to edit.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-08:
could you link the mentioned forum posts?

FEA-eng, 2024-04-08:
> could you link the mentioned forum posts?

There has been quite a lot of them but here are the selected ones:

https://forum.freecad.org/viewtopic.php?t=77387
https://forum.freecad.org/viewtopic.php?t=12742
https://forum.freecad.org/viewtopic.php?t=71620
https://forum.freecad.org/viewtopic.php?t=85900

maxwxyz, 2024-04-09:
@FEA-eng should it work on individual features (like the pattern tool, e.g. bend just one addition of a body) or work on the entire body (current tip position)?

FEA-eng, 2024-04-09:
> I want to work on this issue. Please Assign this to me. Thanks

That would be great, I'm glad you're interested. It can be quite complex though. Unless OCC offers some functionality for that. SheetMetal wb can handle this for its objects after all. The AddFoldWall tool's code can be a good starting point.

> @FEA-eng should it work on individual features (like the pattern tool, e.g. bend just one addition of a body or a selection of features) or work on the entire body (current tip position)?

Ideally, it should work like just another parametric PD operation in the tree taking the previous operation as input but I wouldn't mind the other approach (applying it to the entire body, probably without parametric behavior) either.

maxwxyz, 2024-04-10:
@FEA-eng no what I meant:
1. bend tool works like a default Part Design feature adding on top of the current state of the body. So take the entire body up to that feature and bend it.
2. bend tool works like a Part Design pattern tool: select one or more features in the active body and just bend these (e.g. just select a pad and bend the pad, without bending the other features of the body).

Both cases would be parametric but work in a different way.

FEA-eng, 2024-04-10:
@maxwxyz I think that the first one would be better but it's really hard to say how this could be implemented in FreeCAD. Maybe it will even turn out that the only way is to add it as some non-parametric Part feature to be applied once on the entire model.