FreeCAD/FreeCAD #13362: Twisting existing parts

FEA-eng, 2024-04-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In addition to bending existing parts (#13361), it would be nice to have an option to twist them.

SW example:

[image]

Here FEM could be a workaround too but torsion is currently much more tricky to simulate than bending in FreeCAD FEM.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2026-01-21:
> So the approach to solve this would be to get the normal(s) of the selected plane , add them and rotate (the selected faces) along the calculated normal plane ??

I don't know how it's done internally in SolidWorks, but the idea is that the whole part is twisted between the bounding ("trim") planes. The center of the twist is where you place the orange triad. From there, it propagates to the trim planes as visible in the screenshot attached to this issue and examples shown in the SolidWorks documentation: https://help.solidworks.com/2026/english/SolidWorks/sldworks/HIDD_DVE_FEAT_BENDING.htm?id=22.6.23.1

FlachyJoe, 2026-02-12:
FYI @tomate44 and me are working on this and #13361.

[image]

FlachyJoe, 2026-02-13:
```
gp_Pnt twistAlongX(gp_Pnt from){
    auto pitch = 400.0;
    auto alpha = 4.0 * atan(1.0) / pitch;
    auto x = from.X() ;
    auto y = from.Y() * cos(alpha * from.X()) - from.Z() * sin(alpha * from.X());
    auto z = from.Y() * sin(alpha * from.X()) + from.Z() * cos(alpha * from.X());

    return gp_Pnt(x, y, z);
}
```

The code currently run as a standalone utility, out of FreeCAD. Not ready to share yet.

luzpaz, 2026-03-22:
@mwganson any interest ?