FreeCAD/FreeCAD #13363: Material: Colors get corrupted with imported STEP files after any operation

maxwxyz, 2024-04-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When loading imported STEP files, material and colors look right. After any operation (not on the imported model, just add a Part Cube or new, blank Part Design body) the colors of the imported STEP file are corrupted and each face seems to get a random color. 
This is also non reversible.
If I close the corrupted document without saving and opening it in the same session, the colors are still not the same as the saved state but also different to the last corrupted version, a FreeCAD restart will also not help. If I load the old saved file later, the colors on this STEP solid are broken.
It happens with all step files, created by FreeCAD and other applications.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/b6b6c462-9822-4b21-bc03-6f59c6fb2893

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * OpticsWorkbench 1.0.15
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-04-20:
This is supposed to be fixed with 0a5a1a5c32c9. Please check again with the latest weekly build.

maxwxyz, 2024-04-20:
@wwmayer yes, it seems to work in the latest one, thank you!

closed 2024-04-20 by maxwxyz