FreeCAD/FreeCAD #13364: Gmsh throws "list index out of range" error when a shape is below the FEM Analysis container

FEA-eng, 2024-04-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As mentioned in #13360, Netgen works fine when a shape to be meshed is located below the Analysis container in the tree while Gmsh fails in such cases with the following error message: `list index out of range`

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-04-09:
@FEA-eng, I can't reproduce it. Could you provide the steps?

FEA-eng, 2024-04-09:
> @FEA-eng, I can't reproduce it. Could you provide the steps?

It seems that it may depend on what's in the tree. Try this:
- add Part Cube
- add the Analysis container and Gmsh mesh, material, fixed BC and force load anywhere - everything works so far
- add Part Cylinder and make Part Cut on both shapes (select the cube first) - now geometry will be below the Analysis container
- delete Gmsh mesh and try adding a new one - it should fail with that error while Netgen will still work

If it still doesn't happen for you then it might be OS-specific.

[Gmsh_failure.zip](https://github.com/FreeCAD/FreeCAD/files/14924093/Gmsh_failure.zip)

marioalexis84, 2024-04-09:
I guess you have mesh groups for reference shapes enabled in Fem preferences.
I had already noticed that problem. I have it on my to-do list.

FEA-eng, 2024-04-10:
> I guess you have mesh groups for reference shapes enabled in Fem preferences. I had already noticed that problem. I have it on my to-do list.

Yes, I do. And it seems that it works without it. Good to know that you already have it on the radar.

marioalexis84, 2025-01-25:
@FEA-eng  can you still reproduce this? I can't reproduce it even with mesh groups enabled

FEA-eng, 2025-01-26:
> [@FEA-eng](https://github.com/FEA-eng) can you still reproduce this? I can't reproduce it even with mesh groups enabled

I can't reproduce it now either. I wonder if something could have fixed it. There are only the following errors when the shape is changed:
```
14:30:53  Error: The shapes for the mesh group for the reference shapes of analysis member: ConstraintFixed could not be found!
14:30:53  Error: The shapes for the mesh group for the reference shapes of analysis member: ConstraintForce could not be found!
```