FreeCAD/FreeCAD #13365: Tree reordering doesn't work when a geometry object is below the Analysis container

FEA-eng, 2024-04-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As described in #13360, it is possible to drag and drop a geometry object (like a Part primitive, boolean or PD Body) below the FEM Analysis container but it's not possible to place it back at the top.

[image]

@PaddleStroke FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-07-22:
@FEA-eng can you test again? I can't reproduce.

FEA-eng, 2024-10-26:
> @FEA-eng Can you still reproduce this?

Yes, it's still happening:

[image]

FEA-eng, 2024-11-15:
It turned out to be a general issue (not related only to FEM so @maxwxyz can you remove the FEM label?). The same thing happens e.g. with the TechDraw page container. Moreover, folding the container makes it work.

maxwxyz, 2024-11-15:
@PaddleStroke did this work before?

PaddleStroke, 2024-11-18:
I can't tell. Perhaps it's a corner case that is not handled correctly.

FEA-eng, 2026-04-05:
[This](https://github.com/FreeCAD/FreeCAD/issues/13365#issuecomment-3335013841) is still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260401 (Git shallow)
Build date: 2026/04/01 04:43:57
Build type: Release
Branch: (HEAD detached at 33dcd31f5)
Hash: 33dcd31f5e2175e7dcdfd3caf1d9d86440919bfc
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-10-13:
[This](https://github.com/FreeCAD/FreeCAD/issues/13365#issuecomment-3335013841) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261006 (Git shallow)
Build date: 2026/10/06 21:00:40
Build type: Release
Branch: (HEAD detached at 3424c4106)
Hash: 3424c410627263394f701aeec5b61d7bab7bb3a1
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```