FreeCAD/FreeCAD #13366: Sketcher: box selecting a construction line temporarily changes its appearance

FEA-eng, 2024-04-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I box select a construction line, it changes its appearance (becomes solid white instead of dashed blue). Deselecting it doesn't help but moving its endpoints does.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-09:
@PaddleStroke FYI

FEA-eng, 2024-04-09:
Also, is the highlight color supposed to be black or is it just temporary due to Material stuff or something?

PaddleStroke, 2024-04-09:
I cannot reproduce locally but I have not rebased on master yet. Perhaps its the point color PR, though I'm not sure why it would do that.

luzpaz, 2024-09-08:
@FEA-eng :point_up_2:

FEA-eng, 2024-09-16:
I still have this behavior in the newest weekly. What's interesting, it doesn't seem to happen for lines created as construction geometry, only for those turned into such geometry with Sketcher_ToggleConstruction.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

maxwxyz, 2024-09-16:
@kadet1090 @AjinkyaDahale FYI

AjinkyaDahale, 2024-09-17:
This also happens if a box is made without selecting the curve in question.

maxwxyz, 2024-09-17:
Is it just changing the display style of the geometry or changing it from real to construction? If so, I'd consider this a blocker.

AjinkyaDahale, 2024-09-17:
> Is it just changing the display style of the geometry or changing it from real to construction? If so, I'd consider this a blocker.

It's the former. Safe enough to be called a "known issue" rather than blocker.

AjinkyaDahale, 2024-09-20:
Cannot reproduce on:
```
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 +44 (Git)
Build type: Release
Branch: sk-reduce-cognitive-complexity
Hash: f7008529121288892d687e3d1fe6cd2a1f0d6268
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.8.2.dev
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Curves 0.6.47
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```

benj5378, 2024-10-06:
Having spent multiple hours on this now, I will give up for now.

What I can tell though is, that the color when box selecting isn't set by `EditModeGeometryCoinManager::updateGeometryColor` nor `EditModeCoinManager::updateColor`. I really have no idea where the color or line style is set

luzpaz, 2024-10-06:
@realthunder any chance you can weigh-in on this  ?

benj5378, 2024-10-06:
Sometimes the line even goes black. Could seem like diffuseColor is set in Coin somewhere with an empty variable...?

maxwxyz, 2024-10-06:
Could this be related to the issue https://github.com/FreeCAD/FreeCAD/issues/17028 ? (Although this issue is on Windows)

benj5378, 2024-10-06:
> Could this be related to the issue #17028 ? (Although this issue is on Windows)

On Linux*

Unlikely. I can reproduce this issue, but not #17028. So seems unrelated.

FEA-eng, 2025-06-22:
Still relevant

closed 2025-08-25 by chennes