FreeCAD/FreeCAD #13371: Save dialog - OSX

boboxx, 2024-04-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On the same dialog, the "Don't Save" button is too small in FreeCAD. This is not a issue with the Ondsel version
[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Woodworking 0.21.2.33771
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

azfreed, 2024-05-09:
This appears to be specific to the newer "macOS 14.4.1" as the issue does not appear to occur on macOS 13. I will update to macOS 14 and verify via the following steps:

1. Create a file called "BasicShape.FCstd"
2. After making a small change, close the window without saving.
3. This should bring up the dialog box asking if I want to save without closing.

JohnOCFII, 2024-05-12:
What theme or preference pack are you using?  No issues here using OpenDark from the OpenTheme Add-On.

[image]

```
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.5.3
```

luzpaz, 2024-07-21:
@boboxx @azfreed :point_up_2:

boboxx, 2024-07-22:
Seems to be resolved on my end. How sure at what build number it was ressolved but all the text on button are displayed correctly. This issue can be closed on my end.

luzpaz, 2024-07-24:
Lots of commits between your original revision and current main branch. So unsure, unless some bit-dissects.   
Thanks to all involved. Closing. Please re-open if issue persists.

ryanthara, 2025-11-06:
A new discussion was started in the forum [here](https://forum.freecad.org/viewtopic.php?t=100984). @chennes gives the right hints there. Maybe some other dialogs in FreeCAD has the same issues?

The same behavior on Ubuntu was shown here in [issue #23607](https://github.com/FreeCAD/FreeCAD/issues/23607).