FreeCAD/FreeCAD #13375: Link Array objects in 3D produce wrong dimension in TechDraw

phm87fc, 2024-04-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello together, 

I created an assembly using Link Arrays (Draft OrthoArray and Draft PolarArray). I created a drawing and used the 3d dimensions. 

In the wiki of the _Selection View_ it is described that every feature in the 3D model has a unique combination of `document#Parent.Child`.  See [https://wiki.freecad.org/Selection_view](url). TechDraw does not use this unique identifications and therefore cannot deal with dimensions of link arrays. 

You can find a description of the issue and a confirmation of a second user in the forum:
[https://forum.freecad.org/viewtopic.php?p=751545#p751545](url)
[https://forum.freecad.org/viewtopic.php?p=751640#p751640](url)

Best regards, phm87fc

### Full version info

```shell
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git) AppImage
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Installed mods: 
  * A2plus 0.4.63
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

phm87fc, 2024-04-15:
Hello together,

I finally documented the bug once more:
[Bug_Report_version2.pdf](https://github.com/FreeCAD/FreeCAD/files/14984423/Bug_Report_version2.pdf)

WandererFan, 2024-05-19:
root issue: No globalPlacement() equivalent for LinkElement #13978

note mentions of Spreadsheet, Part and Fastener Wbs.

WandererFan, 2024-06-10:
see also #11977.

maxwxyz, 2024-07-01:
@WandererFan should we keep the 1.0 label?

WandererFan, 2024-09-16:
This will not make it into v1.0.

WandererFan, 2024-12-20:
Finally got this in a shape to submit a PR (#18641).  This is another one where work started and stopped multiple times, so if somebody else were to give it a spin that would be great.

commit abb50a4d (2024-12-23): [TD]Long and link dim refs (fix #13375) (#18641)

* [Meas]Changes for TD dimension refs for links

* [TD]App changes for dim refs to links

* [TD]Gui changes for dim refs to links

* [TD]fix 2 lint messages

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

---------
closed 2024-12-23 by WandererFan