FreeCAD/FreeCAD #13388: Assembly: Add path mate connection/relation

maxwxyz, 2024-04-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be great if the assembly could support a path mate / connection to create a relation between a given path/wire/curve in 3D space and a point or LCS of an object in the assembly.
Depending on the position input (percentage or arc length) the point would move along the path.
@PaddleStroke FYI
[image]
https://help.solidworks.com/2021/English/SolidWorks/sldworks/t_Path_Mate_SWassy.htm?id=d5dac8fe0a4f46b7903fcd1972e6aa9e#Pg0

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-04-10:
Aik-Siong told me that the Ondsel Solver has such a joint. I will need to investigate how to implement it in the wb.

maxwxyz, 2024-04-10:
here is an icon for that:
[image]

maxwxyz, 2024-04-10:
maybe create a new LCS and select a path. Then either let the origin of this LCS move along the path with a percentage or arc length to define position along that path. Optionally specify an axis of the LCS to be tangentially aligned to the position of the origin of the LCS on the path.