FreeCAD/FreeCAD #13389: Assembly: Linear Mate Connection

maxwxyz, 2024-04-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The assembly WB is missing a mate type for a linear relation between two joints / LCS.
E.g. coupling two linear pistons or think of a drawer slide where the outer slide moves double the length of the inner one.
[image]

@PaddleStroke FYI

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-10:
@PaddleStroke here is the icon:
[image]

aiksiongkoh, 2024-06-22:
@maxwxyz What is the actual mechanism that gives the linear relationship? You should model that. I can imagine the two rails to be connected by two rack-pinion joints. The pinions share the same part but have radius r for green rail and 2r for pink rail.

maxwxyz, 2024-06-22:
@aiksiongkoh understand, but modeling two rack and pinion joint for e.g. a hydraulics application is not easy to understand for a beginner, and you need a rotating thing? Or sliders from a drawer which IRL use balls inside of them (https://www.youtube.com/watch?v=8v5SvdPsPQw)
Here is the connector in Onshape as Assembly is modeled after that: https://cad.onshape.com/help/Content/linear_relation.htm?tocpath=Assemblies%7CRelations%7C_____4