FreeCAD/FreeCAD #13390: Assembly: Define Contact between two parts

maxwxyz, 2024-04-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be great if the assembly solver could handle contact situations between multiple bodies. So either end the movement at contact (no intersection of bodies) or drive other bodies by moving one body. Think of a gearset where a driven gear rotates the next gear. With a plain ratio connecting both gears, there is no backlash if you change the rotation direction. A propper surface contact would solve this.
@PaddleStroke FYI

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-04-10:
What you ask for is collision detection basically if I'm not mistaken ?
If so I think this is fairly difficult to implement. There might already be an issue about this btw

maxwxyz, 2024-04-10:
Yes collision detection. I know this might be a long shot and computationally difficult.
It's mentioned here too: https://github.com/FreeCAD/FreeCAD/issues/12445 but this should be an own issue.
Also mentioned here: https://github.com/FreeCAD/FreeCAD/issues/12454 but just for analysis / inspection: check if bodies intersect or have a certain range of clearance and display that. This one here is meant as a type of collision constraint as you mentioned.