FreeCAD/FreeCAD #13392: [Feature Request] ISO style for sketch constraints

NXTler, 2024-04-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello all,
recently [Arc length constraint](https://github.com/FreeCAD/FreeCAD/pull/12602) has been merged into main with ISO style dimensioning, my request is to redo the other constraints to also follow the ISO standard accordingly:
[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git) AppImage
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.4.2
  * A2plus 0.4.63
  * QuickMeasure 2022.10.28
  * Curves 0.6.23
  * CfdOF 1.25.2
  * IconThemes
  * fasteners 0.5.12
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-04-10:
Do you want ISO oriented or ISO referencing? Maybe both with an option?

NXTler, 2024-04-10:
I personally would like to have both, though an option would be better for other preferences.

maxwxyz, 2024-04-10:
@NXTler could you also check TechDraw if they are implemented correctly there?

NXTler, 2024-04-10:
I don't have much experience with FreeCAD's TechDraw workbench and I haven't made a commercial technical drawing in the past few months, but it seems like it's implemented correctly.