FreeCAD/FreeCAD #13393: System menu disables cut, copy, paste on spreadsheet (enabled in context menu though)

hooperbloob, 2024-04-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While the context/popup menu has the cut, copy, and paste options enabled in the Spreadsheet, the same functions in the system menu are disabled - at least on MacOS:

[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-10:
I guess same as in Sketcher WB: there are special commands for copying these items. Maybe they can be triggered with the default cut, copy & paste.

M4xTheMan, 2024-04-12:
CTRL + X / C / V is working in the Spreadsheet WB. Therefore I think the user should find these commands in the toolbar too (as long as they're present there at least) = not grayed out.

kaiwas, 2024-04-13:
> While the context/popup menu has the cut, copy, and paste options enabled in the Spreadsheet, the same functions in the system Edit menu are disabled

another place where similar commands affect different levels.
It looks like the drop-down menu acts on the “whole object” (for example, a spreadsheet object (entirely)).
And inside the table, manipulation is performed on a cell that is located “inside” the table object.

I fully support your opinion that this causes misunderstanding.

luzpaz, 2024-08-18:
Solving this would be nice for new users who are learning to use the Spreadsheet (which we will see an influx of when 1.0 is released). Makes the UI/UX more streamlined. @kadet1090 would you have a moment to diagnose this ?

xtemp09, 2024-12-10:
@luzpaz, I cannot reproduce this issue on _KUbuntu_ and _Windows_. The commit 6178c795817e7f513c9ed1f5673565a4c17fdc5b is supposed to fix it.

hooperbloob, 2024-12-10:
Just tried it on the latest dev release, and while it has improved, the system Edit menu isn't fully functional:

With a newly opened spreadsheet, no cells selected, the Cut, Copy, Delete, and Duplicate selection choices are enabled when they shouldn't be.

Select All doesn't work   -- Cmd-A from within the spreadsheet does work though

When cells are selected in the spreadsheet the system edit menu options work properly.

OS: macOS 15.1.1
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * BIM021 2021.12.0
  * fasteners 0.5.24
  * sheetmetal 0.4.12
  * Assembly4 0.50.15

xtemp09, 2024-12-11:
> With a newly opened spreadsheet,

When you create a spreadsheet, the spreadsheet is selected. If you copy and paste it, you will see.

> Select All doesn't work -- Cmd-A from within the spreadsheet does work though

If you _Select All_, you will select items in the project. You can see that if you open a few spreadsheets and only then choose _Select All_. _Delete_ deletes an item, it does not delete a cell.

Is it the right time to close this issue?

maxwxyz, 2026-11-07:
The remaining Edit-menu behavior is covered in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/6ea2c6c6b7db053aa48b3cd1b88690592f7c6b55

Select All and Delete act on cells in the focused spreadsheet, and Cut or Copy checks for a cell selection. This builds on the earlier upstream copy/paste fix.