FreeCAD/FreeCAD #13402: Arch WB - Crash in ArchAxis with Label Offset dropdown

marcuspollio, 2024-04-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi there,

When accessing the Label Offset dropdown in View Properties Tab of an ArchAxis object, FreeCAD crashes.
To reproduce:

1) Launch FreeCAD, in the Start Page choose BIM/Architecture
2) Add an Axis object (AX shortcut). Select it (either Model Tree or selection in the 3D view).
3) Go to the View Tab of the Properties Panel. In the Axis section, find Label Offset (about in the middle).
4) You can open its sub-properties just fine, but clicking in the values field just next to Label Offset and then on the [...] button on its right leads to a crash. See picture below.

[image]

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS based distro (DE: GNOME)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36811 (Git)
Build type: Unknown
Branch: main
Hash: c0d8de527adb9d68716df092e2fff497ae29d5ea
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
```

### Subproject(s) affected?

Arch

### Anything else?

<details><summary>Crash log</summary>
<p>

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7d10bac42520]
#1  0x7d10c001065f in Gui::ExpressionBinding::bind(App::ObjectIdentifier const&) from /home/USER/build/lib/libFreeCADGui.so+0x57
#2  0x7d10c071cbc3 in Gui::ExpressionSpinBox::bind(App::ObjectIdentifier const&) from /home/USER/build/lib/libFreeCADGui.so+0x27
#3  0x7d10c0714cbd in Gui::QuantitySpinBox::bind(App::ObjectIdentifier const&) from /home/USER/build/lib/libFreeCADGui.so+0x2b
#4  0x7d10c01b56ed in Gui::Dialog::Placement::bindObject() from /home/USER/build/lib/libFreeCADGui.so+0x13d
#5  0x7d10c01b81c4 in Gui::Dialog::TaskPlacement::bindObject() from /home/USER/build/lib/libFreeCADGui.so+0x20
#6  0x7d10c04bb966 in Gui::PropertyEditor::PlacementEditor::browse() from /home/USER/build/lib/libFreeCADGui.so+0x234
#7  0x7d10c07558b8 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::LabelButton::*)()>::call(void (Gui::LabelButton::*)(), Gui::LabelButton*, void**) from /home/USER/build/lib/libFreeCADGui.so+0x7e
#8  0x7d10c0754d50 in void QtPrivate::FunctionPointer<void (Gui::LabelButton::*)()>::call<QtPrivate::List<>, void>(void (Gui::LabelButton::*)(), Gui::LabelButton*, void**) from /home/USER/build/lib/libFreeCADGui.so+0x46
#9  0x7d10c0753a33 in QtPrivate::QSlotObject<void (Gui::LabelButton::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) from /home/USER/build/lib/libFreeCADGui.so+0x84
#10  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f1793) [0x7d10bb6f1793]
#11  0x7d10bc462116 in QAbstractButton::clicked(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#12  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26239e) [0x7d10bc46239e]
#13  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x263fc4) [0x7d10bc463fc4]
#14  0x7d10bc4641e7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#15  0x7d10bc3af4ee in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#16  0x7d10bc36c713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#17  0x7d10bc374364 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#18  0x7d10c003651e in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/USER/build/lib/libFreeCADGui.so+0x114
#19  0x7d10bb6b9e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#20  0x7d10bc372e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#21  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c8d40) [0x7d10bc3c8d40]
#22  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cbfd5) [0x7d10bc3cbfd5]
#23  0x7d10bc36c713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#24  0x7d10c003651e in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/USER/build/lib/libFreeCADGui.so+0x114
#25  0x7d10bb6b9e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#26  0x7d10bbb41307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#27  0x7d10bbb16a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#28  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7d10b54c7d6e]
#29  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7d10bab1bd3b]
#30  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7d10bab71258]
#31  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7d10bab193e3]
#32  0x7d10bb7130b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#33  0x7d10bb6b875b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#34  0x7d10bb6c0cf4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#35  /home/USER/build/lib/libFreeCADGui.so(+0x12cb42f) [0x7d10bfecb42f]
#36  /home/USER/build/lib/libFreeCADGui.so(+0x12cb760) [0x7d10bfecb760]
#37  0x7d10bfecbaa5 in Gui::Application::runApplication() from /home/USER/build/lib/libFreeCADGui.so+0x225
#38  ./FreeCAD(+0x2ddf2) [0x564959421df2]
#39  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7d10bac29d90]
#40  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7d10bac29e40]
#41  ./FreeCAD(+0x2cfc5) [0x564959420fc5]

</p>
</details> 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marcuspollio, 2024-04-11:
This report was initially about the Axis Tool, but after more testing it occurs also in BuildingPart, Building object and Level (Floor) Display Offset properties.

maxwxyz, 2024-04-11:
Is it only on Arch tools or will they also crash for you with other WB features?
Do you have panel overlays enabled, if so which setting?

xtemp09, 2024-04-11:
Can reproduce on 

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Debug
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

<details><summary>Backtrace</summary>

```cpp

ASSERT: "prop != nullptr" in file /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/ExpressionBinding.cpp, line 101
Process 1678 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
    frame #0: 0x00007ffff16969fc libc.so.6`__GI___pthread_kill at pthread_kill.c:44:76
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff16969fc libc.so.6`__GI___pthread_kill at pthread_kill.c:44:76
    frame #1: 0x00007ffff16969b0 libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_internal(signo=6, threadid=140737241193088) at pthread_kill.c:78:10
    frame #2: 0x00007ffff16969b0 libc.so.6`__GI___pthread_kill(threadid=140737241193088, signo=6) at pthread_kill.c:89:10
    frame #3: 0x00007ffff1642476 libc.so.6`__GI_raise(sig=6) at raise.c:26:13
    frame #4: 0x00007ffff16287f3 libc.so.6`__GI_abort at abort.c:79:7
    frame #5: 0x00007ffff65da248 libFreeCADGui.so`messageHandler(type=QtFatalMsg, context=0x00007fffffffb788, msg=0x00007fffffffb688) at Application.cpp:1792:14
    frame #6: 0x00007ffff2097c8f libQt5Core.so.5`qt_message_print(msgType=QtFatalMsg, context=0x00007fffffffb788, message=0x00007fffffffb688) at qlogging.cpp:1843:57
    frame #7: 0x00007ffff2097f3d libQt5Core.so.5`qt_message(msgType=QtFatalMsg, context=0x00007fffffffb788, msg="", ap=0x00007fffffffb690) at qlogging.cpp:378:21
    frame #8: 0x00007ffff209858c libQt5Core.so.5`QMessageLogger::fatal(this=0x00007fffffffb788, msg="") const at qlogging.cpp:890:5
    frame #9: 0x00007ffff20901a7 libQt5Core.so.5`qt_assert(assertion=<unavailable>, file=<unavailable>, line=<unavailable>) at qglobal.cpp:3377:46
    frame #10: 0x00007ffff6728102 libFreeCADGui.so`Gui::ExpressionBinding::bind(this=0x0000555558b71d50, _path=0x00007fffffffbaa0) at ExpressionBinding.cpp:101:5
    frame #11: 0x00007ffff6e8ce21 libFreeCADGui.so`Gui::ExpressionSpinBox::bind(this=0x0000555558b71d50, _path=0x00007fffffffbaa0) at SpinBox.cpp:64:28
    frame #12: 0x00007ffff6e84cc7 libFreeCADGui.so`Gui::QuantitySpinBox::bind(this=0x0000555558b71d20, _path=0x00007fffffffbaa0) at QuantitySpinBox.cpp:299:28
    frame #13: 0x00007ffff68e34a6 libFreeCADGui.so`Gui::Dialog::Placement::bindObject(this=0x00005555597d24b0) at Placement.cpp:789:23
    frame #14: 0x00007ffff68e5f26 libFreeCADGui.so`Gui::Dialog::TaskPlacement::bindObject(this=0x0000555559c06eb0) at Placement.cpp:1044:23
    frame #15: 0x00007ffff6c10b83 libFreeCADGui.so`Gui::PropertyEditor::PlacementEditor::browse(this=0x0000555559be50a0) at PropertyItem.cpp:2463:21
    frame #16: 0x00007ffff6ec9380 libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::LabelButton::*)()>::call(f=0x000000000000000000000000000001a9, o=0x0000555559be50a0, arg=0x00007fffffffbe80) at qobjectdefs_impl.h:152:20
    frame #17: 0x00007ffff6ec8600 libFreeCADGui.so`void QtPrivate::FunctionPointer<void (Gui::LabelButton::*)()>::call<QtPrivate::List<>, void>(f=0x000000000000000000000000000001a9, o=0x0000555559be50a0, arg=0x00007fffffffbe80) at qobjectdefs_impl.h:185:95
    frame #18: 0x00007ffff6ec7137 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::LabelButton::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x0000555555cba440, r=0x0000555559be50a0, a=0x00007fffffffbe80, ret=0x0000000000000000) at qobjectdefs_impl.h:418:49
    frame #19: 0x00007ffff22e33db libQt5Core.so.5`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(a=0x00007fffffffbe80, r=0x0000555559be50a0, this=0x0000555555cba440) at qobjectdefs_impl.h:398:57
    frame #20: 0x00007ffff22e33bf libQt5Core.so.5`void doActivate<false>(sender=0x000055555974dc40, signal_index=9, argv=0x00007fffffffbe80) at qobject.cpp:3925:30
    frame #21: 0x00007ffff22da3ab libQt5Core.so.5`QMetaObject::activate(sender=0x000055555974dc40, m=0x0000000000721000, local_signal_index=2, argv=0x00007fffffffbe80) at qobject.cpp:3985:26
    frame #22: 0x00007ffff2ffaf1a libQt5Widgets.so.5`QAbstractButton::clicked(this=0x000055555974dc40, _t1=<unavailable>) at moc_qabstractbutton.cpp:308:26
    frame #23: 0x00007ffff2ffb22a libQt5Widgets.so.5`QAbstractButtonPrivate::emitClicked(this=0x000055555956acf0) at qabstractbutton.cpp:415:20
    frame #24: 0x00007ffff2ffcbb3 libQt5Widgets.so.5`QAbstractButtonPrivate::click(this=0x000055555956acf0) at qabstractbutton.cpp:408:20
    frame #25: 0x00007ffff2ffcdfe libQt5Widgets.so.5`QAbstractButton::mouseReleaseEvent(this=0x000055555974dc40, e=0x00007fffffffc6a0) at qabstractbutton.cpp:1044:17
    frame #26: 0x00007ffff2f2d919 libQt5Widgets.so.5`QWidget::event(this=0x000055555974dc40, event=0x00007fffffffc6a0) at qwidget.cpp:8671:26
    frame #27: 0x00007ffff2ffe214 libQt5Widgets.so.5`QAbstractButton::event(this=0x000055555974dc40, e=0x00007fffffffc6a0) at qabstractbutton.cpp:1001:26
    frame #28: 0x00007ffff30b83d3 libQt5Widgets.so.5`QPushButton::event(this=0x000055555974dc40, e=0x00007fffffffc6a0) at qpushbutton.cpp:718:34
[... cut ...]
    frame #60: 0x00007ffff1629d90 libc.so.6`__libc_start_call_main(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98) at libc_start_call_main.h:58:16
    frame #61: 0x00007ffff1629e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #62: 0x000055555557eca5 FreeCAD`_start + 37
(lldb)

```

</details>

----

> `Build type: Unknown`

What is that?

marcuspollio, 2024-04-11:
> Is it only on Arch tools or will they also crash for you with other WB features?

It looks like it's mainly related to the Label Offset (src/Mod/Arch/ArchAxis.py L:262-263 and L:516-533) and the Display Offset (src/Mod/Arch/ArchBuildingPart.py L:557-559 and L:761-768).

Other [...] expand buttons (generally in the Data Properties Tab like Placement) work fine. Well, I haven't tested them all =D

> Do you have panel overlays enabled, if so which setting?

Yes in the screenshot above : Toggle Overlay and Transparent. But starting from a new config (without any overlay) also crashes. It does not look related, but cannot say for sure.

Roy-043, 2024-04-11:
The issue occurs since version 0.19.
On Windows there is a warning but no crash:
```
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

I think this is related to Core rather than Arch.
Simple code to create an object that displays the problem:
```
obj = FreeCAD.ActiveDocument.addObject("App::GeometryPython","BuildingPart")
vobj = obj.ViewObject
vobj.addProperty("App::PropertyPlacement","DisplayOffset","BuildingPart","A transformation to apply to the level mark")
```

marcuspollio, 2024-04-11:
> The issue occurs since version 0.19. 

Is there another issue or a forum post about it ?
I tried testing previous versions, but for some reasons both 0.19.3 and 0.18.4 AppImages crash on my system (0.19 on Startup and 0.18 on File>New).
Do you think it's worth trying to bisect ?

> I think this is related to Core rather than Arch.

Do you know of any "similar" cases I could test ? I looked at the main WBs (Part, PD, Draft, TechDraw) but found none.

Roy-043, 2024-04-12:
> Is there another issue or a forum post about it ?

Not to my knowledge.

I don't have experience with bisecting.

What is special about the objects that demonstrate the problem is that they have a Placement property attached to their ViewObject. I suspect that this a rare, and I would look there for the cause.

Maybe @wwmayer has a clue?

marcuspollio, 2024-04-17:
@wwmayer Thank you for having a look !

Anything I can do to help/test ?

wwmayer, 2024-04-17:
Fixed with https://github.com/FreeCAD/FreeCAD/pull/13506

closed 2024-04-17 by wwmayer