FreeCAD/FreeCAD #13412: Can't open STL file with Part module

PeterBan11, 2024-04-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can't open STL file with Part module. My code:

import FreeCAD
import Part

self.path = "back_leg5.STL"
[back_leg5.zip](https://github.com/FreeCAD/FreeCAD/files/14948840/back_leg5.zip)

self.shape = Part.Shape()
print(self.path)
self.shape.read(self.path)
self.boundBox = self.shape.BoundBox
print(self.boundBox)

### Full version info

```shell
FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-11:
can you please provide full about info? Are you using FreeCAD headless?

M4xTheMan, 2024-04-12:
Below you can find details regarding the scripting. Is this what you're looking for? https://wiki.freecad.org/Mesh_FromPartShape

Roy-043, 2024-04-12:
This is indeed something that belongs in the forum. `shape.read` cannot read STL files:

```
>>> help(shape.read)
Help on built-in function read:

read(...) method of Part.Shape instance
    Read in an IGES, STEP or BREP file.
    read(filename)
```

Code that works:
```
import Mesh
import Part

path = "C:/Downloads/back_leg5.STL"

mesh = Mesh.Mesh()
mesh.read(path)
shape = Part.Shape()
shape.makeShapeFromMesh(mesh.Topology, 0.1)
Part.show(shape)
```

See also https://wiki.freecad.org/Part_ShapeFromMesh

closed 2024-04-12 by Roy-043