FreeCAD/FreeCAD #13417: Mesh Repair Tools are fragile

maxwxyz, 2024-04-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Mesh repair tools are more fragile in FreeCAD than other environments.**

Discussion:
I remember having a lot of issues importing simple meshes from the thing that I want to manipulate and then when I try to use the repair tools it complaints.

Please add more context if you have the same issue or ideas for improvements.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-04-12:
Is such a ticket of any help? Its description is unspecific and does not describe a concrete use case.

maxwxyz, 2024-04-12:
I'm ok with closing as not planned, but we said to migrate the issues mentioned on the FreeCAD Day 2024 Complaint Session.

closed 2024-04-12 by maxwxyz